zephyrproject-rtos/zephyr #12051: LOG_PANIC never returns if RTT is disconnected

pdunaj, 2018-12-12:
**Describe the bug**
When RTT is connected LOG_PANIC ends at some point. When RTT is disconnected it does not.

**To Reproduce**
Steps to reproduce the behavior:
Use LOG_PANIC with JLink disconnected.

**Expected behavior**
LOG_PANIC should never lock.

**Impact**
LOG_PANIC deadlocks so application cannot finish its activities.

**Environment (please complete the following information):**
- rev `5ea8454e46208ab3fd7006e98b47ca15d5c7a450`

pdunaj, 2018-12-12:
Hi @nordic-krch , do you know this issue?

pdunaj, 2018-12-13:
@carlescufi I am not sure this is priority low. If panic does not work we will not be able to print messages before immediate reset and this is the place where log messages are, if not most, then very useful (e.g. to indicate reason of failure).

nordic-krch, 2018-12-13:
@pdunaj it's related to the fact that rtt log backend is missing detection for rtt host presence (like rtt console has). I has to be added there. I will try to attack this but probably not today.

pdunaj, 2018-12-13:
Hi @nordic-krch , I have pushed change with a fix if you can have a look.

PR #12087 (2018-12-19): logging: Refactor RTT backend to handle host absence
Refactored blocking operation of RTT backend. Log_output buffer
is used directly in case of blocking operation. Detection of
host presence added. Prevent use of RTT lock when in panic mode.
Removed pending on data being read by the host on every message.

Refactor gives savings of 300 bytes in ROM and 120 bytes in RAM.

Fixes #12051.
PR #12853 (2019-01-30): logging: log_backend_rtt: Fix panic behavior when no host
RTT log backend was locking in panic if host absense was not yet
detected (early panic). It is fixed by adding detection of host
absense while pending on data being read by the host.

Fixes #12051.
closed 2019-01-30 by nashif