zephyrproject-rtos/zephyr #12066: Observing a Python path Issue with Git based Compile

vikramuk, 2018-12-13:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

carlescufi, 2018-12-13:
@vikramuk can you please describe what the issue is?

vikramuk, 2018-12-13:
I know it might be a configuration issue.. but just want someone to update if there have been similar issues during start-up.

SebastianBoe, 2018-12-13:
I don't think using 'bash' (MSYS) on Windows is supported. Can you reproduce the issue with cmd.exe?

carlescufi, 2018-12-13:
@SebastianBoe `bash` is indeed not supported, only `cmd.exe`.

@vikramuk if you want to discuss the relevant forums are the mailing lists and Slack channels. Unless you have a clear description please close this issue and ask for help in the places mentioned.

vikramuk, 2018-12-14:
Hi,
  I still see the Error after running Zephyr-env.cmd

C:\ProgApps\Zephyr\samples\hello_world>cmake -GNinja -DBOARD=stm32_min_dev
CMake Error at C:/ProgApps/cmake-3.13.1-win64-x64/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PythonInterp: Found unsuitable version "1.4", but required
  is at least "3.4" (found C:/ProgApps/Anaconda/python.exe)
Call Stack (most recent call first):
  C:/ProgApps/cmake-3.13.1-win64-x64/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:376 (_FPHSA_FAILURE_MESSAGE)
  C:/ProgApps/Zephyr/cmake/backports/FindPythonInterp.cmake:165 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  C:/ProgApps/Zephyr/cmake/app/boilerplate.cmake:95 (include)
  CMakeLists.txt:3 (include)

-- Configuring incomplete, errors occurred!

SebastianBoe, 2018-12-14:
could you run "C:> where python" ? And "--version" on each of the programs found.

And do the same for 'python3'.

We need to prove that it should be possible for FindPythonInterp.cmake to find an appropriate version.

vikramuk, 2018-12-14:
I ran the same with Qemu [..](https://docs.zephyrproject.org/1.13.0/boards/x86/qemu_x86/doc/board.html)
 Directory of C:\ProgApps\Zephyr\samples\synchronization

12/06/2018  04:47 PM    <DIR>          .
12/06/2018  04:47 PM    <DIR>          ..
12/06/2018  04:47 PM               175 CMakeLists.txt
12/06/2018  04:47 PM                75 prj.conf
12/06/2018  04:47 PM             1,015 README.rst
12/06/2018  04:47 PM               696 sample.yaml
12/06/2018  04:47 PM    <DIR>          src
               4 File(s)          1,961 bytes
               3 Dir(s)  435,901,440,000 bytes free

C:\ProgApps\Zephyr\samples\synchronization>mkdir build

C:\ProgApps\Zephyr\samples\synchronization>cmake -GNinja -DBOARD=qemu_x86
-- Found PythonInterp: C:/ProgApps/Py37/python.exe (found suitable version "3.7.1", minimum required is "3.4")
CMake Error at C:/ProgApps/Zephyr/cmake/app/boilerplate.cmake:102 (message):
  Source directory equals build directory.  In-source builds are not
  supported.  Please specify a build directory, e.g.  cmake -Bbuild -H.
Call Stack (most recent call first):
  CMakeLists.txt:3 (include)

-- Configuring incomplete, errors occurred!

C:\ProgApps\Zephyr\samples\synchronization>where python
C:\ProgApps\Py37\python.exe

PR #12095 (2018-12-21): cmake: Fixed python detection bug
Fixed a bug where an old version of python on path would break the
python detection mechanism.

This fixes #12066
commit 56d144cd (2018-12-21): cmake: Fixed python detection bug

Fixed a bug where an old version of python on path would break the
python detection mechanism.

This fixes #12066
closed 2018-12-21 by carlescufi