zephyrproject-rtos/zephyr #12069: iwdg_stm32.c: potential failure in wdt_set_config()

vaussard, 2018-12-13:
I noticed one potential issue in iwdg_stm32_set_config() while reading the code.

**Describe the bug**

The function iwdg_stm32_set_config() is busy waiting while the IWDG is not ready (https://elixir.bootlin.com/zephyr/latest/source/drivers/watchdog/iwdg_stm32.c#L108).

There is a safety net, as it will return -ENODEV after IWDG_DEFAULT_TIMEOUT. But the comment of IWDG_DEFAULT_TIMEOUT is wrong, since it says "48 ms" (https://elixir.bootlin.com/zephyr/latest/source/drivers/watchdog/iwdg_stm32.c#L30) but k_uptime_get_32() returns hardware cycles. Thus we compare apples with oranges, and we might return -ENODEV after only 48 cycles (while we expected to wait for 48 ms before failing).

Apparently this is not happening and the watchdog initializes fast enough (at least in my case), but this is a potential issue. Either the comment is wrong, or the code is wrong.

**To Reproduce**
This is only based on code review, the issues went unnoticed as it is very hard/unlikely to trigger. Maybe it also depends on the configuration of the clocks.

**Impact**
Failure to configure the watchdog timeout.

erwango, 2018-12-14:
This driver will be deprecated when moving to new counter API.
Let's check this issue is taken into account when implementing the new driver (https://github.com/zephyrproject-rtos/zephyr/pull/11401).

erwango, 2019-02-01:
@vaussard, can you double check, it seems that k_uptime_get_32 actually returns ms (cf https://github.com/zephyrproject-rtos/zephyr/blob/c39bb49e8458fd6548711ee01bd1fafe4ffa2e95/kernel/timeout.c#L238)

erwango, 2019-02-12:
@vaussard , is that ok to close this issue according to above comment?

vaussard, 2019-02-21:
Hi @erwango, sorry for slow answering. Yes indeed, I do not know what I smoked that day, this is returning ms. I suppose that I mixed up with k_cycle_get_32(), shame on me. Let's close.

closed 2019-02-21 by vaussard