zephyrproject-rtos/zephyr #12088: RFC: Device definitions

derchrismakaio, 2018-12-14:
### TL;DR
**Please show me a way to simplify device assignments in DTS** 

### Details
Dear All,

sorry for raising an issue here, especially because I've already sent this to the mailing list, but I think it is more structured here (formatting, etc.).

I'm currently creating a driver for the max170xx fuel gauge family and I like to create a nice OOBE.

Here's what I did so far (concerning DTS).

Created a **max170xx.yaml**:

```
---
title: MAX170XX fuel gauge family
version: 0.1

description: >
    This is a representation of the MAX170XX fuel gauge family.

inherits:
  !include i2c-device.yaml

properties:
    compatible:
      type: string
      category: required
      constraint: "max170xx"
      generation: define
...
```

add the device to our board's dts at i2c1 (board is pca10056):

```
&i2c1 {
	status = "ok";
    sda-pin = <24>;
    scl-pin = <41>;

    max170xx@0 {
        label = "max17043";
        compatible = "max170xx";
        status = "ok";
        reg = <0x36 >;
    };
};
```

In reg property i defined the max17043's slave address (which is different from max17055's slave address).

This creates the following in **generated_dts_board.h**:
```
#define I2C_1_MAX170XX_0_BASE_ADDRESS				DT_NORDIC_NRF_I2C_40004000_MAX170XX_0_BASE_ADDRESS
#define I2C_1_MAX170XX_0_BUS_NAME				DT_NORDIC_NRF_I2C_40004000_MAX170XX_0_BUS_NAME
#define I2C_1_MAX170XX_0_LABEL					DT_NORDIC_NRF_I2C_40004000_MAX170XX_0_LABEL
```

Now I nearly have a board-independent definition which I can use in my driver's max170xx.h:
```
#ifdef I2C_0_MAX170XX_0_BUS_NAME
#define MAX170XX_I2C_BUS_NAME I2C_0_MAX170XX_0_BUS_NAME
#define MAX170XX_I2C_DEV_NAME I2C_0_MAX170XX_0_LABEL
#define MAX170XX_I2C_ADDRESS I2C_0_MAX170XX_0_BASE_ADDRESS
#elif defined(I2C_1_MAX170XX_0_BUS_NAME)
#define MAX170XX_I2C_BUS_NAME I2C_1_MAX170XX_0_BUS_NAME
#define MAX170XX_I2C_DEV_NAME I2C_1_MAX170XX_0_LABEL
#define MAX170XX_I2C_ADDRESS I2C_1_MAX170XX_0_BASE_ADDRESS
#endif
``` 

Here are the caveats: 
- it currently checks only for i2c_0 and i2c_1, it'd rather have something like I2C_MAX170XX_BUS_NAME created
- so I changed max170xx@0 to max170xx and removed reg (slave address could be statically defined in device header) 
- but then the closest generated define is I2C_1_MAX170XX_40004000_MAX170XX_LABEL which now contains the board specific i2c address, it seems I need the unnecessary max170xx**@0** 

I know I could create a dts_fixup but that means a dts_fixup for every board. Given that I want to add like 5 drivers and that I'd be happy to support all boards, that would mean a lot of work which - in my naive point of view - seems so superfluous.

Finally the questions:
- is it okay proceeding with  max170xx@0-syntax or will it mess with something?
- is there any other option to create board-independent device assignments? (Hopefully I made clear what I wish for by now)

erwango, 2018-12-14:
Hi @derchrismakaio ,

This is a valid question, and I think there is no perfect solution available right now, but some people are currently looking into it.
Maybe you can look at https://github.com/zephyrproject-rtos/zephyr/pull/11977 which might be a solution to your problem.

galak, 2019-01-08:
> * is it okay proceeding with  max170xx@0-syntax or will it mess with something?

No, this should be max170xx@36

galak, 2019-01-08:
There are some enhancements that in progress to address this issue.  Going to close this as a duplicate and lets focus on the already existing issues for further discussion.

closed 2019-01-08 by galak