zephyrproject-rtos/zephyr #12102: Use -DSHIELD= to enable building of a shield

erwango, 2018-12-14:
**Is your enhancement proposal related to a problem? Please describe.**
Current way of selecting a shield is to enable matching Kconfig symbol, cf https://github.com/zephyrproject-rtos/zephyr/blob/bd34448a7576c57d6212981c0f187c50a69f7c0c/samples/shields/x_nucleo_iks01a1/prj.conf#L2
This is no more possible if were going to run DT before Kconfig .

**Describe the solution you'd like**
We need an alternate way to select a shield.
The only option that I can think of right now is to implement a new -DSHIELD= build option to enable selection and build of a shield.

**Describe alternatives you've considered**
NA

**Additional context**
NA

erwango, 2018-12-14:
@SebastianBoe, would you have time to check this ?

SebastianBoe, 2018-12-14:
I agree with this approach.

SebastianBoe, 2018-12-14:
When the set of CMake variables and their inter-related dependencies grows in size we might want to consider creating a Kconfig tree specifically for DT configuration.

But for now CMake variables will have to do.

erwango, 2018-12-18:
@galak , I'm working on #12092 which is more tricky than expected.
Will probably not be able to address this point before next year

PR #12403 (2019-01-18): cmake: dts: move to specifying shield on the command line
Rather than specifying SHIELD via Kconfig, we move it to being specified
via the command line, similar to board.

Fixes #12102
closed 2019-01-18 by galak