zephyrproject-rtos/zephyr #12109: Combine duplicate kinetis watchdog early boot init code

MaureenHelm, 2018-12-14:
**Is your enhancement proposal related to a problem? Please describe.**
Kinetis K64 and KW2xD SoCs have duplicate watchdog init code

**Describe the solution you'd like**
Combine the duplicate code so it only exists in one place. Consider moving it from assembly into C.

**Describe alternatives you've considered**
The watchdog must be initialized or disabled within the first 256 bus clock cycles after reset; the normal device driver init `mcux_wdog_init` is too late.

**Additional context**
This proposal was suggested in #11815 and #11852

henrikbrixandersen, 2019-04-22:
Would it possible to just call `WDOG32_Deinit()` from `fsl_wdog32.c` in the MCUxpresso SDK?

DerekSnell, 2025-02-21:
Closing this Enhancement request, it is out of date.

Currently, these watchdogs are disabled in the HAL `SystemInit()` startup code, which is called by each SOC in the `soc_reset_hook()`.

|                     | K64F | KW2xD |
| ------------- | ------------- | -- |
| soc_reset_hook | [k6x/soc.c](https://github.com/zephyrproject-rtos/zephyr/blob/efbe277d203f8d25f11818662abad6ea0316bfa4/soc/nxp/kinetis/k6x/soc.c#L168C6-L168C20) | [soc_kw2xd.c](https://github.com/zephyrproject-rtos/zephyr/blob/efbe277d203f8d25f11818662abad6ea0316bfa4/soc/nxp/kinetis/kwx/soc_kw2xd.c#L161) |
| HAL SystemInit()  | [system_MK64F12.c](https://github.com/zephyrproject-rtos/hal_nxp/blob/3fc5f811fb8c0bdd90a8d965c50707172ce80d05/mcux/mcux-sdk/devices/MK64F12/system_MK64F12.c#L102)  | [system_MKW24D5.c](https://github.com/zephyrproject-rtos/hal_nxp/blob/3fc5f811fb8c0bdd90a8d965c50707172ce80d05/mcux/mcux-sdk/devices/MKW24D5/system_MKW24D5.c#L119) |