zephyrproject-rtos/zephyr #12114: assertion using nRF5 power clock with BLE and nRF5 temp sensor

dhananjaygj, 2018-12-14:
**Describe the bug**
When nRF5 temperature sensor is used during BLE communication, assertion is hit in `temp_nrf5_sample_fetch()` with `-EBUSY` error (_temp_nrf5.c_ line 54). Assertion is hit during `clock_control_off()` since both BLE controller and nRF Temperature uses the same nRF power clock. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create sample BLE peripheral application with nRF temperature sensor. 
2. Poll the temperature sensor every 2 seconds.
3. Connect the sample BLE application with any peer
4. At some point assertion is hit as explained above.

**Expected behavior**
Assertion should not occur or if it occurs API documentation can give more information  

**Impact**
Even if this issue is minor this affects just the Debug build of our application and release build works fine. Hopefully there is no problem using BLE along with nRF temperature sensor in parallel.

**Environment (please complete the following information):**
 - Zephyr
 - nRF52840 board

dhananjaygj, 2018-12-14:
Below is part of the discussion with @anangl regarding this observation 

> ok, so the assertion fails when the clock is released
then I think the assertion in line 54 is incorrect
I am not sure, because the clock_control API does not precise what error codes should `clock_control_off()` return
and from what I see in the code, the nrf5 clock_control returns `-EBUSY` when there is another module that requested the clock and still needs it, so the clock cannot be turned off at the moment
so only the reference counter is decremented and the clock will be turned off after the last module that previously requested it releases it as well (edited)
therefore, I think what you are observing is a pretty normal situation, the clock is no longer needed for the temperature sensor, but it cannot be just turned off because it is still needed for BLE
so the `-EBUSY` error is not actually signaling that something is wrong, it is rather an information that the request of releasing  the clock was accepted, but the clock itself cannot be turned of yet

dhananjaygj, 2018-12-17:
I submitted a patch based on suggestions from @anangl here https://github.com/zephyrproject-rtos/zephyr/pull/12133

nashif, 2019-04-02:
@anangl ping

PR #17293 (2019-10-04): drivers: clock_control: nrf driver reimplementation
Clock_control has been re-implemented to support new clock control API and calibration algorithm which optionally uses temperature sensor and skips calibration on no temperature change.

- Added new API implementation
- Added calibration algorithm (calibration support extracted to separate file)
- Added test suite
- Updated usage of clock _control API in nordic drivers, removed hacky usage of subsys parameter.

-----------
Fixes #12114
closed 2019-10-04 by carlescufi