zephyrproject-rtos/zephyr #12118: printf doesn't work on qemu_cortex_m3 with newlib and arm gcc 2018q2 toolchain

madandylau, 2018-12-15:
**Describe the bug**
Take the sample "external_lib" for example. It crashes with newlib and arm gcc toolchain. But it works well with the zephyr-sdk 0.9.5.

**To Reproduce**
1. source zephyr-env.sh, export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb and export GNUARMEMB_TOOLCHAIN_PATH to my arm gcc toolchain path
2. cd samples/application_development/external_lib && mkdir build
3. cmake -GNinja -DBOARD=qemu_cortex_m3 ..
4. ninja menuconfig (C Library -> check the "Build with newlib c library") 
5. ninja
6. ninja run

**Screenshots or console output**
```
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
qemu: fatal: Lockup: can't escalate 3 to HardFault (current priority -1)

R00=0000002a R01=20000638 R02=00009cb3 R03=00000000
R04=20000658 R05=00000000 R06=000010cd R07=20000638
R08=00000000 R09=ffffffff R10=00000000 R11=00000000
R12=00000800 R13=200005ec R14=000011af R15=00000000
XPSR=00000003 ---- A handler
FPSCR: 00000000
Aborted (core dumped)
FAILED: zephyr/CMakeFiles/run
cd /home/liu/zephyr/samples/application_development/external_lib/build && /usr/bin/qemu-system-arm -cpu cortex-m3 -machine lm3s6965evb -nographic -vga none -pidfile qemu.pid -serial mon:stdio -kernel /home/liu/zephyr/samples/application_development/external_lib/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: gcc-arm-none-eabi-7-2018-q2-update
- Zephyr version: 1.13.0

**Additional context**
This problem also happens on my stm32f769i_disco target.

aurel32, 2018-12-15:
@madandylau If it works with the zephyr sdk but not with another toolchain, it's likely that's the wrong version of newlib is linked, probably the non-thumb version. If I am correct, using either the current git version of Zephyr or the patches from PR #11641 should fix the issue.

madandylau, 2018-12-16:
It's exactly the problem. Solved with the current git version. Thanks @aurel32

closed 2018-12-16 by madandylau