zephyrproject-rtos/zephyr #12120: Can't pair with nrf52832 running zephyr

M1cha, 2018-12-15:
**Describe the bug**
I bought an nrf52832 on aliexpress and am trying to run zephyr on it using the "nrf52_pca10040" config.
schecmatics: https://ae01.alicdn.com/kf/HTB1kysEdhrI8KJjy0Fpq6z5hVXav.jpg?size=85239&height=783&width=1000&hash=374338387ec77f380a36ef738c15d7d8

A key difference is that the clock is connected to XC1/XC2 which is called `Synth` in the nrf52 sdk.
That's why I have to change `NRFX_CLOCK_CONFIG_LF_SRC`  to `2` to successfully run the nordic examples.

For zephyr I've done something similar by adding support for the synth clock(this diff also includes my config changes so you can see exactly what I did):
```
diff --git a/drivers/clock_control/Kconfig.nrf5 b/drivers/clock_control/Kconfig.nrf5
index 506f33d9cf..fecc932157 100644
--- a/drivers/clock_control/Kconfig.nrf5
+++ b/drivers/clock_control/Kconfig.nrf5
@@ -38,6 +38,9 @@ config CLOCK_CONTROL_NRF5_K32SRC_RC
 config CLOCK_CONTROL_NRF5_K32SRC_XTAL
 	bool "Crystal Oscillator"
 
+config CLOCK_CONTROL_NRF5_K32SRC_SYNTH
+	bool "Synth"
+
 endchoice
 
 config CLOCK_CONTROL_NRF5_K32SRC_BLOCKING
diff --git a/include/drivers/clock_control/nrf5_clock_control.h b/include/drivers/clock_control/nrf5_clock_control.h
index 5c64ac522c..7778598978 100644
--- a/include/drivers/clock_control/nrf5_clock_control.h
+++ b/include/drivers/clock_control/nrf5_clock_control.h
@@ -20,6 +20,9 @@
 #ifdef CONFIG_CLOCK_CONTROL_NRF5_K32SRC_XTAL
 #define CLOCK_CONTROL_NRF5_K32SRC 1
 #endif
+#ifdef CONFIG_CLOCK_CONTROL_NRF5_K32SRC_SYNTH
+#define CLOCK_CONTROL_NRF5_K32SRC 2
+#endif
 
 /* Define 32KHz clock accuracy */
 #ifdef CONFIG_CLOCK_CONTROL_NRF5_K32SRC_500PPM
diff --git a/samples/bluetooth/peripheral/prj.conf b/samples/bluetooth/peripheral/prj.conf
index 8735e7fa64..944823b01b 100644
--- a/samples/bluetooth/peripheral/prj.conf
+++ b/samples/bluetooth/peripheral/prj.conf
@@ -21,3 +21,10 @@ CONFIG_FLASH_MAP=y
 CONFIG_FCB=y
 CONFIG_SETTINGS=y
 CONFIG_SETTINGS_FCB=y
+
+CONFIG_HAS_SEGGER_RTT=y
+CONFIG_USE_SEGGER_RTT=y
+CONFIG_RTT_CONSOLE=y
+CONFIG_UART_CONSOLE=n
+
+CONFIG_CLOCK_CONTROL_NRF5_K32SRC_SYNTH=y
```

This way I can at least run zephyr and discover the device.
The problem is that I can't start a pairing because zepyhr disconnects after a few seconds:
```
***** Booting Zephyr OS zephyr-v1.13.0-2782-g07eb5d1eaa *****
Bluetooth initialized
Advertising successfully startedcore: HW Platform: Nordic Semiconductor (0x0002)

[00:00:00.001,892] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.001,892] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 1.13 Build 99
[00:00:00.002,166] <wrn> bt_hci_core: No ID address. Expecting one to come from storage.
[00:00:00.011,108] <inf> bt_hci_core: Identity: c5:bd:4f:cd:3a:4f (random)
[00:00:00.011,108] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0x05f1
Connected.011,108] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0xffff
Disconnected (reason 8)
```

**Expected behavior**
The device should successfully pair.

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: gcc version 8.2.0 (Arch Repository)
 - Commit SHA: 07eb5d1eaa

M1cha, 2018-12-17:
@cvinayak thanks for the tip, that seems to work.
What's the proper way to do this though? enabling the HF clock from my main function and never disabling it doesn't sound very power-efficient.

cvinayak, 2018-12-18:
@M1cha i just looked at schematic, you "shall" use LF RC, not synth. Its not very useful to use synth unless there are other requirements to keep HF clock always on.

M1cha, 2018-12-22:
you're totally right, using RC works just fine.

closed 2018-12-22 by M1cha