zephyrproject-rtos/zephyr #12123: samples: cmsis_rtos_v1/philosophers faults with a FATAL EXCEPTION on esp32

nniranjhana, 2018-12-17:
**Describe the bug**
The test in samples/portability/cmsis_rtos_v1/philosophers runs into a FATAL EXCEPTION on esp32 platforms.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=esp32
3. make
4. See error

**Screenshots or console output**

 ```
***** Booting Zephyr OS zephyr-v1.13.0-2782-g07eb5d1eaa (delayed boot 1000ms) *****
 [2J[15;1HDemo Description
 ----------------
 An implementation of a solution to the Dining Philosophers
 problem (a classic multi-thread synchronization problem) using
 CMSIS RTOS V1 APIs. This particular implementation demonstrates the
 usage of multiple preemptible threads of differing
 priorities, as well as semaphores and thread sleeping.
 [6;1HPhilosopher 5 [P: 3]        STARVING
 [6;1HPhilosopher 5 [P: 3]    HOLDING ONE FORK
  ** FATAL EXCEPTION
  ** CPU 0 EXCCAUSE 28 PS 0x00060920 PC 0x40082e1b VADDR 0x00000064
  **  A0 0x80080f9a  SP 0x3ffb4060  A2 0x3ffb1e54  A3 0x00000006
  **  A4 0x00000001  A5 0x00000000  A6 0x00000003  A7 0x00000000
  **  A8 0x3ff
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.9.5

spoorthik, 2019-01-16:
I am not able to reproduce the issue on the latest commit:

Commit: 81ec05548dbafb602d2d400ef86ffcf2e1732c31
```
***** Booting Zephyr OS zephyr-v1.13.0-3237-g81ec055 (delayed boot 1000ms) *****
Philosopher 0 [C:-2]  THINKING [  300 ms ]
Philosopher 1 [C:-1]  THINKING [  625 ms ]
Philosopher 2 [P: 0]   EATING  [  275 ms ]
Philosopher 4 [P: 2]   EATING  [  475 ms ]
Philosopher 5 [P: 3]  THINKING [  725 ms ]

Demo Description
----------------
An implementation of a solution to the Dining Philosophers
problem (a classic multi-thread synchronization problem) using
CMSIS RTOS V1 APIs. This particular implementation demonstrates the
usage of multiple preemptible threads of differing
priorities, as well as semaphores and thread sleeping.
```

rgundi, 2019-02-18:
Since this is not reproducible on the latest HEAD, suggest closing it.

ManojSubbarao, 2019-02-26:
Issue cannot be reproduced as per @rgundi. Closing the issue.

closed 2019-02-26 by ManojSubbarao