zephyrproject-rtos/zephyr #12129: Include: Clean up header file namespace

yashi, 2018-12-17:
**Is your feature request related to a problem? Please describe.**
Because Zephyr places a few header files under `include/`,  they could conflict application authors header files.

**Describe the solution you'd like**
Move all Zephyr header files under `include/zephyr/` and make other directories specified by `zephyr_include_directories()` not to interfere with application headers.  (I haven't checked all directories yet)

**Describe alternatives you've considered**
Tell users to namespace their header file.  That could be a solution but if users are using existing codebase, it's tedious to do that.

**Additional context**
There was a thread on dev list.
https://lists.zephyrproject.org/g/devel/topic/namespacing_of_zephyr_headers/16760352

pfalcon, 2018-12-17:
I can only +1 this, as I raised this issue almost 2 years ago in the thread linked in the description.

cc: @nashif, @galak, @carlescufi, @MaureenHelm and other TSC members.

pabigot, 2018-12-18:
Any chance this could get done before 1.14?

carlescufi, 2018-12-18:
@pabigot I have now marked it TSC, so it will be reviewed during the next call

galak, 2019-01-16:
Is the intent that applications would do:

`#include <zephyr/foo.h>`

nashif, 2019-01-16:
TSC: moved to dev-review to be reviewed in dev meeting

galak, 2019-01-17:
@yashi are you able to make these changes?

yashi, 2019-01-18:
Sure.  I can work on it.

Does it mean we have a go? I'm not sure what the conclusion is in dev-review.

In the simple way, we can just do `$ mv include/* include/zephyr/` and fix errors we have.
But it's gonna be a huge patch, so I'm thinking about moving a few files in a commit.  Let me know if you have better way to do this kind of work.

@pabigot seems to want this change before 1.14.  Are we aiming the next release? Do we have candidate date for the release?

SebastianBoe, 2019-01-18:
IMHO the pain from breaking all out-of-tree "#include " statements does not outweigh the pain from resolving the occasional aliasing issue.

pfalcon, 2019-01-18:
> IMHO the pain from breaking all out-of-tree "#include " statements does not outweigh the pain from resolving the occasional aliasing issue.

There wouldn't be pain if the baseline best practices ("don't think that your project is the only one in the universe or don't think that your project is the center of the universe") were followed from the beginning. There would be less pain if it was done when raised a year go. Longer the wait, bigger the pain. And yep, pain is the price of not doing it right, and learning to do it right.

pfalcon, 2019-01-18:
@yashi:

> But it's gonna be a huge patch, so I'm thinking about moving a few files in a commit.

That's what I would suggest, prepare an initial commit with a few files moved, and wait for feedback. If they say put more stuff in the same PR, do it (I'd personally prefer that it moved gradually, but yep, targeting 1.14, or it'll be way more pain overall).

pabigot, 2019-01-24:
@yashi please proceed with a sample with a couple headers and we'll see what it looks like.  Thanks.  (From dev review call.)

yashi, 2019-01-27:
OK.

for each `include/*.h`, we have the following number of files including:
```
$ cd zephyr/
$ for i in include/*.h; do echo $(git grep -l "#include [<\"]$(basename $i)[>\"]" | wc -l): $i; done | sort -n
0: include/crc16.h
0: include/crc32.h
0: include/crc8.h
1: include/gna.h
2: include/kernel_version.h
2: include/tty.h
3: include/can.h
3: include/exc_handle.h
3: include/json.h
3: include/sched_priq.h
4: include/aio_comparator.h
4: include/base64.h
4: include/cache.h
4: include/stats.h
5: include/console.h
5: include/irq_nextlevel.h
5: include/shared_irq.h
6: include/ring_buffer.h
6: include/rtc.h
7: include/disk_access.h
7: include/led_strip.h
7: include/ptp_clock.h
8: include/display.h
8: include/flash_map.h
8: include/led.h
8: include/syscall.h
10: include/adc.h
10: include/counter.h
10: include/fcb.h
11: include/spinlock.h
12: include/i2s.h
12: include/watchdog.h
13: include/crc.h
13: include/kernel_includes.h
15: include/dma.h
17: include/entropy.h
17: include/tracing.h
18: include/sys_clock.h
19: include/pwm.h
20: include/ipm.h
22: include/power.h
27: include/sw_isr_table.h
31: include/generated_dts_board.h
33: include/irq.h
41: include/irq_offload.h
43: include/flash.h
46: include/spi.h
54: include/clock_control.h
56: include/atomic.h
56: include/fs.h
66: include/uart.h
91: include/sys_io.h
93: include/pinmux.h
108: include/i2c.h
110: include/sensor.h
158: include/gpio.h
162: include/toolchain.h
393: include/init.h
505: include/device.h
522: include/zephyr.h
530: include/kernel.h
```
I'll start from smaller ones and move to `kernel.h` and other directories.

pabigot, 2019-01-27:
@yashi thanks.  

We'll probably have a long argument about how the changes should be merged (batches of files in each merge, all files in separate commits in one merge, all files in one commit in one merge).

If you want to do it one commit per include file to start with I'd be fine with that, and if you can do it with a script that makes the changes for one file at time that'd also be wise, as it may take a while to get this approved and in the meantime the codebase will keep changing so the PR will have to be updated.

In the past I've done something like:
```
git ls-tree -r --name-only HEAD | xargs sed -i -e "something involving ${name}"
git commit -a -m "passim: put ${name} header into zephyr namespace"
```

pfalcon, 2019-01-31:
Of course, we need to clean up not just header locations, but what headers export: https://github.com/zephyrproject-rtos/zephyr/issues/12935

nashif, 2019-06-01:
This request was voted on and was rejected by the TSC (22 May 2019)

closed 2019-06-01 by nashif