zephyrproject-rtos/zephyr #12130: Is zephyr targeting high-end phone or pc with open ended computation on the roadmap?

lktc420, 2018-12-17:
Is zephyr targeting high-end phone or pc with fast processor, large ram and do open ended computation on the roadmap? like support 64bit system.

I'm evaluating a user case like self-driving car central process system, it will have very powerful
hardware but still require hard real time support. Now most solution would be patch the linux with
some soft real time patch, but if there is a native rtos with FuSA that can do this, it will be very helpful.

carlescufi, 2018-12-17:
@lktc420 well, it depends on what you mean by "high-end phone or pc". Zephyr already supports x86 and will soon support x86-64. When it comes to Arm, Cortex-A is not supported an will probably not be for a while.

lktc420, 2018-12-18:
> @lktc420 well, it depends on what you mean by "high-end phone or pc". Zephyr already supports x86 and will soon support x86-64. When it comes to Arm, Cortex-A is not supported an will probably not be for a while.

Thanks for the reply!
I mean generally is the goal and roadmap of zephyr would concern much on high-end computation. 
For example, like the google zircon rtos kernl claim it self:
https://fuchsia.googlesource.com/zircon/+/master/docs/zx_and_lk.md
"On the other hand, Zircon targets modern phones and modern personal computers with fast processors, non-trivial amounts of ram with arbitrary peripherals doing open ended computation."

I don't know if this modern computer support only related to 64 bit and multi-core support?

nashif, 2019-12-12:
answered.

closed 2019-12-12 by nashif