zephyrproject-rtos/zephyr #12134: I cannot see a Zephyr way to change the clock frequency at runtime

gon1332, 2018-12-17:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to have an abstract way to change the clock settings at runtime. E.g. the frequency, the clock source.

**Describe the solution you'd like**
An API, maybe an enhancement of `clock_control`? Or a proposal, reference if something like this is already supported.

**Describe alternatives you've considered**
Maybe the alternative that you would propose is to integrate such functionality in the SOC-level implementation of power subsystem. But what if I wanted to scale the frequency when not idling?

**Additional context**
-

gon1332, 2018-12-24:
Well, I'm searching on how I can implement some of the Low Power Modes that STM32L4s support on Zephyr RTOS. Some of them, like Low Power Run, mode requires that the system frequency decreases in order to enter to it. Off course a senario like this cannot be implemented as a Zephyr Power Management Policy, because the process doesn't sleep.

I'm just exploring here, and see what fits according to our expectations and needs. (About the `CONFIG_TIMER_READS_ITS_FREQUENCY_AT_RUNTIME` I've already seen it, used it, and had some tests with amperometer, etc.)

Another question is how to change the clock source of the system, or of a specific peripheral. As I see in `clock_control` source, the clock_control_init function only takes the compile time configurations from `board_defconfig` and initializes a clock for the peripheral only once. Is my observation true? If yes, in order to have better control over the system and its capabilities, wouldn't be good to also provide a setter API for the system/peripherals?

andyross, 2018-12-25:
The clock_control API is a driver-level API for controlling clocks, as might be used for peripheral devices (a UART or SPI clock, stuff like that).  It doesn't seem to me to have been designed with CPU frequency scaling in mind, but I don't see why it couldn't be extended to do that.

Honestly I'm not sure the API is very interesting.   Merely setting a clock isn't very hard or complicated, the problem comes from the fact that every platform that does this has its own weird behavior.  To me, that sort of argues against having a generic API at all.

I think I retreat to my previous answer: the spots in the Zephyr kernel that are sensitive to cycle timing are pretty rare, and mostly captured already by that hacky tunable (which was introduced AFAIK for the HPET driver, not strictly for frequency scaling).  My suggestion would be to prototype this as a STM32-specific feature and then worry about how to extend it to other SoC's.

gon1332, 2018-12-26:
So, for now, I'm working like you suggested. I will continue like this but on a PR in order to get feedback as I progress.

erwango, 2019-07-26:
^^ @FRASTM

erwango, 2019-07-26:
^^ @Benichou34

closed 2020-02-13 by nashif