zephyrproject-rtos/zephyr #12144: There's lots of references to undefined Kconfig symbols

ulfalizer, 2018-12-18:
Here's a list of Kconfig symbols that are referenced somewhere but not defined in the (main) Kconfig tree, sorted by name and with the locations where they're referenced. If you see something you recognize, then please submit a PR to remove it. `git grep` is nice for finding references.

There's some false positives in there (like for `checkpatch.pl` and `kconfiglib.py`, and due to token pasting), but I'm just dumping it as is. Once it's been pruned a bit, I'll add some easy whitelisting scheme(s) (could allow undefined symbol names in comments if they're surrounded by single quotes, etc.)

I'm planning to make it an automatic check later.

    CONFIG_ADC_QMSI_SAMPLE_WIDTH        doc/release-notes-1.5.rst:230, doc/subsystems/test/sanitycheck.rst:342
    CONFIG_ADXL372_SPI_GPIO_CS          drivers/sensor/adxl372/adxl372.c:901, drivers/sensor/adxl372/adxl372.c:935, drivers/sensor/adxl372/adxl372.h:286, drivers/sensor/adxl372/adxl372.h:322
    CONFIG_ADXL372_SPI_GPIO_CS_DRV_NAME drivers/sensor/adxl372/adxl372.c:936
    CONFIG_ADXL372_SPI_GPIO_CS_PIN      drivers/sensor/adxl372/adxl372.c:937
    CONFIG_ALTERA_AVALON_EPCS           ext/hal/altera/CMakeLists.txt:84
    CONFIG_ALTERA_AVALON_SGDMA          ext/hal/altera/CMakeLists.txt:52
    CONFIG_ALTERA_AVALON_SPI            ext/hal/altera/CMakeLists.txt:76
    CONFIG_ALTERA_AVALON_UART           ext/hal/altera/CMakeLists.txt:27
    CONFIG_APA102_STRIP_BUS_NAME        drivers/led_strip/apa102.c:84
    CONFIG_APA102_STRIP_FREQUENCY       drivers/led_strip/apa102.c:89
    CONFIG_APA102_STRIP_NAME            drivers/led_strip/apa102.c:103
    CONFIG_ARM_SAU                      arch/arm/core/cortex_m/tz/arm_core_tz.c:138, arch/arm/core/cortex_m/tz/arm_core_tz.c:166, arch/arm/include/cortex_m/tz.h:234, arch/arm/include/cortex_m/tz.h:270
    CONFIG_BMM150_SET_ATTR              doc/release-notes-1.13.rst:595
    CONFIG_BT_6LOWPAN                   samples/bluetooth/ipsp/README.rst:22
    CONFIG_CAN_DEV                      samples/drivers/CAN/src/main.c:194
    CONFIG_CCM                          doc/devices/dts/device_tree.rst:297, scripts/dts/extract/globals.py:26
    CONFIG_CCM_BASE_ADDRESS             include/arch/arm/cortex_m/scripts/linker.ld:96, include/arch/arm/cortex_m/scripts/linker.ld:97, include/arch/arm/cortex_m/scripts/linker.ld:476, include/arch/arm/cortex_m/scripts/linker.ld:513, include/linker/linker-defs.h:255, include/linker/linker-defs.h:265, kernel/init.c:146, kernel/init.c:175
    CONFIG_CCM_SIZE                     include/arch/arm/cortex_m/scripts/linker.ld:97
    CONFIG_CMU                          soc/arm/silabs_exx32/common/soc.c:119
    CONFIG_COMPILER_OPT_AS_LIST         CMakeLists.txt:274, CMakeLists.txt:275
    CONFIG_DCD_DSR                      samples/net/wpan_serial/src/main.c:600
    CONFIG_DESCRIPTION                  scripts/checkpatch.pl:2813
    CONFIG_DIR                          doc/CMakeLists.txt:153, doc/CMakeLists.txt:181, doc/CMakeLists.txt:210
    CONFIG_ESP32_I2C_0_RX_LSB_FIRST     drivers/i2c/i2c_esp32.c:594
    CONFIG_ESP32_I2C_0_TX_LSB_FIRST     drivers/i2c/i2c_esp32.c:592
    CONFIG_ESP32_I2C_1_RX_LSB_FIRST     drivers/i2c/i2c_esp32.c:641
    CONFIG_ESP32_I2C_1_TX_LSB_FIRST     drivers/i2c/i2c_esp32.c:639
    CONFIG_ESP_DIF_LIBRARY              ext/hal/esp/CMakeLists.txt:6
    CONFIG_EXPERIMENTAL                 .checkpatch.conf:14
    CONFIG_FFT                          cmake/extensions.cmake:717, cmake/extensions.cmake:793, cmake/extensions.cmake:803
    CONFIG_FIRQ_NO_LPCC                 arch/arc/core/fast_irq.S:78, arch/arc/core/fast_irq.S:135
    CONFIG_FLASH_BASE_ADDRESS           arch/arm/core/cortex_m/prep_c.c:50, drivers/flash/flash_gecko.c:43, drivers/flash/flash_gecko.c:66, drivers/flash/flash_gecko.c:151, drivers/flash/flash_sam.c:127, drivers/flash/flash_sam.c:216, drivers/flash/flash_sam.c:280, drivers/flash/flash_sam0.c:39, drivers/flash/flash_sam0.c:273, drivers/flash/flash_stm32.c:117, drivers/flash/flash_stm32.c:133, drivers/flash/flash_stm32f0x.c:38, drivers/flash/flash_stm32f0x.c:80, drivers/flash/flash_stm32f3x.c:30, drivers/flash/flash_stm32f3x.c:38, drivers/flash/flash_stm32f3x.c:58, drivers/flash/flash_stm32f3x.c:73, drivers/flash/flash_stm32f4x.c:49, drivers/flash/flash_stm32f7x.c:49, drivers/flash/flash_stm32l4x.c:49, include/arch/arm/cortex_m/scripts/linker.ld:43, samples/mpu/mpu_test/src/main.c:20, scripts/dts/extract/flash.py:69, scripts/meta/west/runners/core.py:385, scripts/meta/west/runners/core.py:386, scripts/meta/west/runners/core.py:392, scripts/meta/west/runners/core.py:395, soc/arc/quark_se_c1000_ss/dts_fixup.h:16, soc/arc/quark_se_c1000_ss/linker.ld:16, soc/arm/arm/beetle/arm_mpu_regions.c:13, soc/arm/arm/mps2/arm_mpu_regions.c:15, soc/arm/atmel_sam/common/arm_mpu_regions.c:15, soc/arm/nordic_nrf/nrf52/mpu_mem_cfg.h:19, soc/arm/nordic_nrf/nrf52/mpu_regions.c:19, soc/arm/nxp_imx/rt/arm_mpu_regions.c:18, soc/arm/nxp_kinetis/k6x/nxp_mpu_regions.c:40, soc/arm/st_stm32/common/arm_mpu_regions.c:18, soc/nios2/nios2-qemu/dts_fixup.h:9, soc/nios2/nios2f-zephyr/dts_fixup.h:9, soc/x86/apollo_lake/dts_fixup.h:11, soc/x86/atom/dts_fixup.h:21, soc/x86/ia32/dts_fixup.h:21, soc/x86/intel_quark/quark_d2000/dts_fixup.h:13, soc/x86/intel_quark/quark_se/dts_fixup.h:17, soc/x86/intel_quark/quark_x1000/dts_fixup.h:19, subsys/usb/class/usb_dfu.c:291, subsys/usb/class/usb_dfu.c:425, subsys/usb/class/usb_dfu.c:620, subsys/usb/class/usb_dfu.c:627
    CONFIG_FLASH_SIZE                   arch/arm/defconfig:6, drivers/flash/flash_gecko.c:141, drivers/flash/flash_sam.c:63, drivers/flash/flash_sam.c:67, drivers/flash/flash_sam.c:314, drivers/flash/flash_sam0.c:52, drivers/flash/flash_sam0.c:77, drivers/flash/flash_sam0.c:319, drivers/flash/flash_stm32f0x.c:159, drivers/flash/flash_stm32f7x.c:78, drivers/flash/flash_stm32f7x.c:82, drivers/flash/flash_stm32f7x.c:86, drivers/flash/flash_stm32f7x.c:173, drivers/flash/flash_stm32f7x.c:180, drivers/flash/flash_stm32f7x.c:187, drivers/flash/soc_flash_mcux.c:115, include/arch/arm/cortex_m/scripts/linker.ld:46, include/arch/arm/cortex_m/scripts/linker.ld:53, scripts/dts/extract/flash.py:70, soc/arc/quark_se_c1000_ss/dts_fixup.h:17, soc/arc/quark_se_c1000_ss/linker.ld:17, soc/arm/atmel_sam/common/arm_mpu_mem_cfg.h:13, soc/arm/atmel_sam/common/arm_mpu_mem_cfg.h:15, soc/arm/atmel_sam/common/arm_mpu_mem_cfg.h:17, soc/arm/atmel_sam/common/arm_mpu_mem_cfg.h:19, soc/arm/atmel_sam/common/arm_mpu_mem_cfg.h:21, soc/arm/atmel_sam/common/arm_mpu_mem_cfg.h:23, soc/arm/atmel_sam/common/arm_mpu_mem_cfg.h:25, soc/arm/nordic_nrf/nrf52/mpu_mem_cfg.h:12, soc/arm/nordic_nrf/nrf52/mpu_mem_cfg.h:14, soc/arm/nordic_nrf/nrf52/mpu_mem_cfg.h:16, soc/arm/nordic_nrf/nrf52/mpu_mem_cfg.h:20, soc/arm/nordic_nrf/nrf52/mpu_mem_cfg.h:22, soc/arm/nordic_nrf/nrf52/mpu_mem_cfg.h:24, soc/arm/nordic_nrf/nrf52/mpu_mem_cfg.h:26, soc/arm/nxp_imx/rt/arm_mpu_mem_cfg.h:13, soc/arm/nxp_imx/rt/arm_mpu_mem_cfg.h:15, soc/arm/nxp_imx/rt/arm_mpu_mem_cfg.h:17, soc/arm/nxp_imx/rt/arm_mpu_mem_cfg.h:19, soc/arm/nxp_imx/rt/arm_mpu_mem_cfg.h:21, soc/arm/nxp_imx/rt/arm_mpu_mem_cfg.h:23, soc/arm/nxp_imx/rt/arm_mpu_mem_cfg.h:25, soc/arm/nxp_imx/rt/arm_mpu_mem_cfg.h:27, soc/arm/st_stm32/common/arm_mpu_mem_cfg.h:13, soc/arm/st_stm32/common/arm_mpu_mem_cfg.h:15, soc/arm/st_stm32/common/arm_mpu_mem_cfg.h:17, soc/arm/st_stm32/common/arm_mpu_mem_cfg.h:19, soc/arm/st_stm32/common/arm_mpu_mem_cfg.h:21, soc/arm/st_stm32/common/arm_mpu_mem_cfg.h:23, soc/arm/st_stm32/common/arm_mpu_mem_cfg.h:25, soc/nios2/nios2-qemu/dts_fixup.h:11, soc/nios2/nios2f-zephyr/dts_fixup.h:11, soc/x86/apollo_lake/dts_fixup.h:15, soc/x86/atom/dts_fixup.h:25, soc/x86/ia32/dts_fixup.h:25, soc/x86/intel_quark/quark_d2000/dts_fixup.h:17, soc/x86/intel_quark/quark_se/dts_fixup.h:21, soc/x86/intel_quark/quark_x1000/dts_fixup.h:23
    CONFIG_FOO                          doc/scripts/genrest.py:2, doc/scripts/genrest.py:22, scripts/kconfig/kconfiglib.py:215, scripts/kconfig/kconfiglib.py:223, scripts/kconfig/kconfiglib.py:1029, scripts/kconfig/kconfiglib.py:1525, scripts/kconfig/kconfiglib.py:1556, scripts/kconfig/kconfiglib.py:1719, scripts/kconfig/kconfiglib.py:1719, scripts/meta/west/runners/core.py:115, scripts/meta/west/runners/core.py:115
    CONFIG_FOO_LOG_LEVEL                doc/subsystems/logging/logger.rst:170, doc/subsystems/logging/logger.rst:183, doc/subsystems/logging/logger.rst:198, doc/subsystems/logging/logger.rst:206, doc/subsystems/logging/logger.rst:243, doc/subsystems/logging/lo [... cut ...]
    CONFIG_USB_DC_STM                   boards/arm/nucleo_f207zg/pinmux.c:41, boards/arm/nucleo_l4r5zi/pinmux.c:62
    CONFIG_USB_HS_BASE_ADDRES           drivers/usb/device/usb_dc_stm32.c:58
    CONFIG_USB_MAX_SPEED                drivers/usb/device/usb_dc_stm32.c:315
    CONFIG_WHATEVER                     doc/subsystems/test/ztest.rst:188, doc/subsystems/test/ztest.rst:249
    CONFIG_WLAN                         samples/net/zperf/src/zperf_shell.c:39
    CONFIG_XX                           doc/release-notes-1.13.rst:656

SebastianBoe, 2018-12-18:
PR submitted for CONFIG_LTO: https://github.com/zephyrproject-rtos/zephyr/pull/12163

galak, 2019-01-15:
Can you provide a script that generates this output so we can update this PR from time to time with the latest results.

galak, 2019-01-29:
> Can you provide a script that generates this output so we can update this PR from time to time with the latest results.

@ulfalizer are you able to provide a script

nashif, 2019-01-30:
```
CONFIG_ALTERA_AVALON_EPCS           ext/hal/altera/CMakeLists.txt:84
CONFIG_ALTERA_AVALON_SGDMA          ext/hal/altera/CMakeLists.txt:52
CONFIG_ALTERA_AVALON_SPI            ext/hal/altera/CMakeLists.txt:76
CONFIG_ALTERA_AVALON_UART           ext/hal/altera/CMakeLists.txt:27
CONFIG_ARM_SAU                      arch/arm/core/cortex_m/tz/arm_core_tz.c:138, arch/arm/core/cortex_m/tz/arm_core_tz.c:166, arch/arm/include/cortex_m/tz.h:234, arch/arm/include/cortex_m/tz.h:270
```

addressed in https://github.com/zephyrproject-rtos/zephyr/pull/12875

ulfalizer, 2019-01-30:
@galak 
Here's the script I used, run from the Zephyr root. I'll clean it up a bit later once I'm done moving.

Will need to have `scripts/kconfig` in `PYTHONPATH` so that kconfiglib can be found. Could also install it with `sudo pip3 kconfiglib` or just copy/symlink it into the Zephyr root.

```python
import collections
import os
import subprocess
import sys

import kconfiglib

os.environ["SOC_DIR"] = "soc/"
os.environ["BOARD_DIR"] = "boards/*/*"
os.environ["ARCH"] = "*"
defined_syms = set(sym.name for sym in kconfiglib.Kconfig().unique_defined_syms)

# gitglossary "magic signature: /"
grep_cmd = ("git", "grep", "--only-matching", "--line-number",
            "--extended-regexp", "--word-regexp", "-I", "--null",
            "CONFIG_[A-Z0-9_]+")

grep_process = subprocess.Popen(grep_cmd, stdout=subprocess.PIPE,
                                stderr=subprocess.PIPE)

stdout, stderr = grep_process.communicate()

if grep_process.returncode:
    # stderr will be 'bytes' here, which prints as b'oops'
    sys.exit("error: '{}' failed with exit code {} and wrote {} to stderr"
             .format(" ".join(grep_cmd), grep_process.returncode, stderr))

stdout = stdout.decode("utf-8")

undef_to_locs = collections.defaultdict(list)

for line in stdout.splitlines():
    path, lineno, sym_name = line.split("\0")
    if sym_name[7:] not in defined_syms and not sym_name.endswith("_"):
        undef_to_locs[sym_name].append("{}:{}".format(path, lineno))

for name in sorted(undef_to_locs):
    print("{:35} {}".format(name, ", ".join(undef_to_locs[name])))
```

galak, 2019-01-30:
Here's an updated list as of commit: b73c9b32e8d84ad0a7820b5087db7cab681114f1

```
CONFIG_ADC_QMSI_SAMPLE_WIDTH        doc/release-notes-1.5.rst:230, doc/subsystems/test/sanitycheck.rst:342
CONFIG_ADXL372_SPI_GPIO_CS          drivers/sensor/adxl372/adxl372.c:901, drivers/sensor/adxl372/adxl372.c:935, drivers/sensor/adxl372/adxl372.h:286, drivers/sensor/adxl372/adxl372.h:322
CONFIG_ADXL372_SPI_GPIO_CS_DRV_NAME drivers/sensor/adxl372/adxl372.c:936
CONFIG_ADXL372_SPI_GPIO_CS_PIN      drivers/sensor/adxl372/adxl372.c:937
CONFIG_ALTERA_AVALON_EPCS           ext/hal/altera/CMakeLists.txt:84
CONFIG_ALTERA_AVALON_SGDMA          ext/hal/altera/CMakeLists.txt:52
CONFIG_ALTERA_AVALON_SPI            ext/hal/altera/CMakeLists.txt:76
CONFIG_ALTERA_AVALON_UART           ext/hal/altera/CMakeLists.txt:27
CONFIG_ARM_SAU                      arch/arm/core/cortex_m/tz/arm_core_tz.c:138, arch/arm/core/cortex_m/tz/arm_core_tz.c:166, arch/arm/include/cortex_m/tz.h:234, arch/arm/include/cortex_m/tz.h:270
CONFIG_BMM150_SET_ATTR              doc/release-notes-1.13.rst:595
CONFIG_BOARD_PCA10090_INTERFACE2_MCU boards/arm/nrf52840_pca10090/board.c:99, boards/arm/nrf52840_pca10090/board.c:102, boards/arm/nrf52840_pca10090/integrity.c:19
CONFIG_BT_6LOWPAN                   samples/bluetooth/ipsp/README.rst:45
CONFIG_BT_ADV_MAX                   subsys/bluetooth/controller/ll_sw/lll.h:25, subsys/bluetooth/controller/ll_sw/lll.h:27, subsys/bluetooth/controller/ll_sw/lll.h:42, subsys/bluetooth/controller/ll_sw/ull_adv.c:65, subsys/bluetooth/controller/ll_sw/ull_adv.c:830, subsys/bluetooth/controller/ll_sw/ull_adv.c:844, subsys/bluetooth/controller/ll_sw/ull_adv.c:996, subsys/bluetooth/shell/ll.c:196, subsys/bluetooth/shell/ll.c:210, subsys/bluetooth/shell/ll.c:222, subsys/bluetooth/shell/ll.c:232, subsys/bluetooth/shell/ll.c:240
CONFIG_BT_CTLR_MAX_CONN             subsys/bluetooth/controller/ll_sw/ull.c:173, subsys/bluetooth/controller/ll_sw/ull.c:175, subsys/bluetooth/controller/ll_sw/ull.c:179
CONFIG_BT_SCAN_MAX                  subsys/bluetooth/controller/ll_sw/ull_scan.c:48, subsys/bluetooth/controller/ll_sw/ull_scan.c:49, subsys/bluetooth/controller/ll_sw/ull_scan.c:98, subsys/bluetooth/controller/ll_sw/ull_scan.c:286
CONFIG_CAN_DEV                      samples/drivers/CAN/src/main.c:194
CONFIG_CCM                          doc/devices/dts/device_tree.rst:297, scripts/dts/extract/globals.py:25
CONFIG_CCM_BASE_ADDRESS             include/arch/arm/cortex_m/scripts/linker.ld:96, include/arch/arm/cortex_m/scripts/linker.ld:97, include/arch/arm/cortex_m/scripts/linker.ld:512, include/arch/arm/cortex_m/scripts/linker.ld:549, include/linker/linker-defs.h:248, include/linker/linker-defs.h:258, kernel/init.c:147, kernel/init.c:180
CONFIG_CCM_SIZE                     include/arch/arm/cortex_m/scripts/linker.ld:97
CONFIG_CMU                          soc/arm/silabs_exx32/common/soc.c:124
CONFIG_COMPILER_OPT_AS_LIST         CMakeLists.txt:292, CMakeLists.txt:293
CONFIG_DCCM_BASE_ADDRESS            boards/arc/iotdk/arc_mpu_regions.c:21, boards/arc/iotdk/arc_mpu_regions.c:50
CONFIG_DCCM_SIZE                    boards/arc/iotdk/arc_mpu_regions.c:22, boards/arc/iotdk/arc_mpu_regions.c:51
CONFIG_DCD_DSR                      samples/net/wpan_serial/src/main.c:600
CONFIG_DEEP_SLEEP                   ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:13, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:16, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:68, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:128, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:162, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:199, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:223, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:250
CONFIG_DESCRIPTION                  scripts/checkpatch.pl:2813
CONFIG_DIR                          doc/CMakeLists.txt:154, doc/CMakeLists.txt:182, doc/CMakeLists.txt:211
CONFIG_DISC_LOG_LEVEL               subsys/disk/disk_access_sdhc.c:7
CONFIG_ESP32_I2C_0_RX_LSB_FIRST     drivers/i2c/i2c_esp32.c:594
CONFIG_ESP32_I2C_0_TX_LSB_FIRST     drivers/i2c/i2c_esp32.c:592
CONFIG_ESP32_I2C_1_RX_LSB_FIRST     drivers/i2c/i2c_esp32.c:641
CONFIG_ESP32_I2C_1_TX_LSB_FIRST     drivers/i2c/i2c_esp32.c:639
CONFIG_ESP_DIF_LIBRARY              ext/hal/esp/CMakeLists.txt:6
CONFIG_EXPERIMENTAL                 .checkpatch.conf:14
CONFIG_FFT                          cmake/extensions.cmake:736, cmake/extensions.cmake:812, cmake/extensions.cmake:822
CONFIG_FIRQ_NO_LPCC                 arch/arc/core/fast_irq.S:78, arch/arc/core/fast_irq.S:135
CONFIG_FLAG                         include/misc/util.h:213, include/misc/util.h:217, include/misc/util.h:217
CONFIG_FLASH_BASE_ADDRESS           arch/arm/core/cortex_m/prep_c.c:50, boards/arc/iotdk/arc_mpu_regions.c:31, boards/arc/iotdk/arc_mpu_regions.c:54, drivers/flash/flash_gecko.c:43, drivers/flash/flash_gecko.c:66, drivers/flash/flash_gecko.c:151, drivers/flash/flash_sam.c:127, drivers/flash/flash_sam.c:216, drivers/flash/flash_sam.c:280, drivers/flash/flash_sam0.c:39, drivers/flash/flash_sam0.c:273, drivers/flash/flash_stm32.c:117, drivers/flash/flash_stm32.c:133, drivers/flash/flash_stm32f0x.c:38, drivers/flash/flash_stm32f0x.c:80, drivers/flash/flash_stm32f3x.c:30, drivers/flash/flash_stm32f3x.c:38, drivers/flash/flash_stm32f3x.c:58, drivers/flash/flash_stm32f3x.c:73, drivers/flash/f [... cut ...]
CONFIG_USB_STM32                    boards/arm/disco_l475_iot1/pinmux.c:49, boards/arm/disco_l475_iot1/pinmux.c:54
CONFIG_WEBSOCKET_CONSOLE            subsys/console/line_fifo.c:24, subsys/console/line_fifo.c:42
CONFIG_WHATEVER                     doc/subsystems/test/ztest.rst:188, doc/subsystems/test/ztest.rst:249
CONFIG_WLAN                         samples/net/zperf/src/zperf_shell.c:39
CONFIG_XUK_64_BIT_ABI               arch/x86_64/core/xuk-config.h:22, arch/x86_64/core/xuk.c:111
CONFIG_XX                           doc/release-notes-1.13.rst:656
```

galak, 2019-01-30:
Deal with CONFIG_MCUX_IGPIO* in PR #12883

carlescufi, 2019-02-14:
@ulfalizer @galak please update the script to whitelist `CONFIG_BT_6LOWPAN`. That one is there because it's a **Linux** Kconfig option.

ulfalizer, 2019-02-15:
@carlescufi 
At some point (probably soon), I will add a CI check and just whitelist everything that remains.

ulfalizer, 2019-02-15:
Here's an updated list:

```
CONFIG_ADC_QMSI_SAMPLE_WIDTH        doc/guides/test/sanitycheck.rst:342, doc/releases/release-notes-1.5.rst:230
CONFIG_BMM150_SET_ATTR              doc/releases/release-notes-1.13.rst:595
CONFIG_BT_6LOWPAN                   samples/bluetooth/ipsp/README.rst:45
CONFIG_CAN_DEV                      samples/drivers/CAN/src/main.c:194
CONFIG_CLOUD_AUDIENCE               samples/net/google_iot_mqtt/src/protocol.c:272
CONFIG_CLOUD_CLIENT_ID              samples/net/google_iot_mqtt/src/protocol.c:47
CONFIG_CLOUD_PUBLISH_TOPIC          samples/net/google_iot_mqtt/src/protocol.c:54
CONFIG_CLOUD_SUBSCRIBE_CONFIG       samples/net/google_iot_mqtt/src/protocol.c:48, samples/net/google_iot_mqtt/src/protocol.c:49, samples/net/google_iot_mqtt/src/protocol.c:139, samples/net/google_iot_mqtt/src/protocol.c:152, samples/net/google_iot_mqtt/src/protocol.c:355
CONFIG_CMU                          soc/arm/silabs_exx32/common/soc.c:124
CONFIG_COMPILER_OPT_AS_LIST         CMakeLists.txt:286, CMakeLists.txt:287
CONFIG_COUNTER_RTC                  drivers/counter/counter_nrfx_rtc.c:249, drivers/counter/counter_nrfx_rtc.c:252, drivers/counter/counter_nrfx_rtc.c:264, drivers/counter/counter_nrfx_rtc.c:272
CONFIG_COUNTER_RTC_STM32_CLOCK_SRC  drivers/counter/counter_ll_stm32_rtc.c:286
CONFIG_COUNTER_TIMER                drivers/counter/counter_nrfx_timer.c:228, drivers/counter/counter_nrfx_timer.c:231, drivers/counter/counter_nrfx_timer.c:249, drivers/counter/counter_nrfx_timer.c:257
CONFIG_DCD_DSR                      samples/net/wpan_serial/src/main.c:600
CONFIG_DEEP_SLEEP                   ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:13, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:16, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:68, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:128, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:162, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:199, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:223, ext/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/fsl_msmc.c:250
CONFIG_DESCRIPTION                  scripts/checkpatch.pl:2812
CONFIG_DIR                          doc/CMakeLists.txt:175, doc/CMakeLists.txt:203, doc/CMakeLists.txt:232
CONFIG_ERR                          ext/hal/microchip/mec/MCHP_MEC1701.h:1520, ext/hal/microchip/mec/MCHP_MEC1701.h:29941, ext/hal/microchip/mec/MCHP_MEC1701.h:29942
CONFIG_ESP_DIF_LIBRARY              ext/hal/esp/CMakeLists.txt:6
CONFIG_ETH_DW_0_IRQ_SHARED_NAME     drivers/ethernet/eth_dw.c:370, soc/x86/intel_quark/quark_x1000/dts_fixup.h:48
CONFIG_EXPERIMENTAL                 .checkpatch.conf:14
CONFIG_FFT                          cmake/extensions.cmake:736, cmake/extensions.cmake:812, cmake/extensions.cmake:822
CONFIG_FIRQ_NO_LPCC                 arch/arc/core/fast_irq.S:78, arch/arc/core/fast_irq.S:135
CONFIG_FLAG                         include/misc/util.h:213, include/misc/util.h:217, include/misc/util.h:217
CONFIG_FOO                          doc/scripts/genrest.py:2, doc/scripts/genrest.py:22, scripts/kconfig/kconfiglib.py:215, scripts/kconfig/kconfiglib.py:223, scripts/kconfig/kconfiglib.py:1043, scripts/kconfig/kconfiglib.py:1558, scripts/kconfig/kconfiglib.py:1589, scripts/kconfig/kconfiglib.py:1752, scripts/kconfig/kconfiglib.py:1752, scripts/west_commands/runners/core.py:115, scripts/west_commands/runners/core.py:115
CONFIG_FOO_LOG_LEVEL                doc/reference/logging/index.rst:174, doc/reference/logging/index.rst:187, doc/reference/logging/index.rst:202, doc/reference/logging/index.rst:210, doc/reference/logging/index.rst:247, doc/reference/logging/index.rst:260, doc/reference/logging/index.rst:274, include/logging/log.h:343, include/logging/log.h:381
CONFIG_FOO_SETTING_1                doc/guides/kconfig/index.rst:602
CONFIG_FOO_SETTING_2                doc/guides/kconfig/index.rst:603
CONFIG_GDB_INFO                     doc/releases/release-notes-1.7.rst:235
CONFIG_GPIO_BUTTON_DEV              samples/drivers/CAN/src/main.c:212
CONFIG_GPIO_DW_0_IRQ_SHARED_NAME    drivers/gpio/gpio_dw.c:505, soc/x86/intel_quark/quark_x1000/dts_fixup.h:44
CONFIG_GPIO_DW_1_IRQ_SHARED_NAME    drivers/gpio/gpio_dw.c:574
CONFIG_GPIO_DW_2_IRQ_SHARED_NAME    drivers/gpio/gpio_dw.c:641
CONFIG_GPIO_DW_3_IRQ_SHARED_NAME    drivers/gpio/gpio_dw.c:708
CONFIG_GPIO_LED_DEV                 samples/drivers/CAN/src/main.c:204
CONFIG_HPET_TIMER_DEBUG             doc/releases/release-notes-1.7.rst:236
CONFIG_I2C_DW_0_IRQ_SHARED_NAME     drivers/i2c/i2c_dw.c:707, soc/x86/intel_quark/quark_x1000/dts_fixup.h:40
CONFIG_I2C_VIRTUAL                  tests/drivers/i2c/i2c_slave_api/common/CMakeLists.txt:1, tests/drivers/i2c/i2c_slave_api/prj_virtual.conf:10
CONFIG_I2C_VIRTUAL_NAME             tests/drivers/i2c/i2c_slave_api/common/i2c_virtual.c:226
CONFIG_IEEE802154_NRF5_RAW          ext/hal/nordic/drivers/CMakeLists.txt:1
CONFIG_ISA_ARM                      include/toolchain/gcc.h:170
CONFIG_ISA_THUMB                    include/toolchain/common.h:67, include/toolchain/gcc.h:153, include/toolchain/gcc.h:296
CONFIG_LEGACY_KERNEL                doc/releases/release-notes-1.7.rst:374, doc/releases/release-notes-1.7.rst:439
CONFIG_LIS2DW12_INT_PIN             drivers/sensor/lis2dw12/lis2dw12.c:358
CONFIG_LIST                         misc/generated/CMakeLists.txt:5, misc/generated/CMakeLists.txt:9
[... cut ...]
CONFIG_UART_QMSI_0_BAUDRATE         boards/x86/arduino_101/arduino_101_mcuboot_defconfig:10
CONFIG_UART_QMSI_1_BAUDRATE         boards/x86/arduino_101/doc/index.rst:446
CONFIG_USB_CONSOLE                  doc/guides/kconfig/index.rst:163, doc/guides/kconfig/index.rst:178
CONFIG_WHATEVER                     doc/guides/test/ztest.rst:188, doc/guides/test/ztest.rst:249
CONFIG_XUK_64_BIT_ABI               arch/x86_64/core/xuk-config.h:22, arch/x86_64/core/xuk.c:111
CONFIG_XX                           doc/releases/release-notes-1.13.rst:656
```

ulfalizer, 2019-02-15:
Pinning to hopefully get rid of a few more. Will dump the rest into a whitelist and add a CI check after that.

Maybe the CI check should be restricted to just source files too.

galak, 2019-02-15:
> Pinning for a while to hopefully get rid of a few more. Will dump the rest into a whitelist and add a CI check after that.
> 
> Maybe the CI check should be restricted to just code too.

Did you update your script?

ulfalizer, 2019-02-15:
@galak 
Hacked up a new version that works. Can get a reference for how to set the environment variables from [scripts/check_compliance.py](https://github.com/zephyrproject-rtos/ci-tools/blob/master/scripts/check_compliance.py) in the ci-tools repo.

```python
import collections
import os
import subprocess
import sys

sys.path.insert(0, "scripts/kconfig")
import kconfiglib

os.environ["SOC_DIR"] = "soc/"
os.environ["ARCH_DIR"] = "arch/"
os.environ["BOARD_DIR"] = "boards/*/*"
os.environ["ARCH"] = "*"
os.environ['GENERATED_DTS_BOARD_CONF'] = "dummy"
defined_syms = set(sym.name for sym in kconfiglib.Kconfig().unique_defined_syms)

# gitglossary "magic signature: /"
grep_cmd = ("git", "grep", "--only-matching", "--line-number",
            "--extended-regexp", "--word-regexp", "-I", "--null",
            "CONFIG_[A-Z0-9_]+")

grep_process = subprocess.Popen(grep_cmd, stdout=subprocess.PIPE,
                                stderr=subprocess.PIPE)

stdout, stderr = grep_process.communicate()

if grep_process.returncode:
    # stderr will be 'bytes' here, which prints as b'oops'
    sys.exit("error: '{}' failed with exit code {} and wrote {} to stderr"
             .format(" ".join(grep_cmd), grep_process.returncode, stderr))

stdout = stdout.decode("utf-8")

undef_to_locs = collections.defaultdict(list)

for line in stdout.splitlines():
    path, lineno, sym_name = line.split("\0")
    if sym_name[7:] not in defined_syms and not sym_name.endswith("_"):
        undef_to_locs[sym_name].append("{}:{}".format(path, lineno))

for name in sorted(undef_to_locs):
    print("{:35} {}".format(name, ", ".join(undef_to_locs[name])))
```

ulfalizer, 2019-02-15:
Ah... maybe you meant re. false positives in `samples/`, etc. Nope, not updated for that yet.

galak, 2019-02-15:
> Ah... maybe you meant re. false positives in `samples/`, etc. Nope, not updated for that yet.

that was one question.  Another is when running:

```
Kconfig.zephyr:43: '/Kconfig.modules' does not exist. Perhaps the $srctree
environment variable (unset or blank) is set incorrectly. Note that the current
value of $srctree is saved when the Kconfig instance is created (for consistency
and to cleanly separate instances).
```

ulfalizer, 2019-02-15:
@galak 
Oh, right, I also removed the `source` at `Kconfig.zephyr:43`. CI just writes a blank file there. It's for including extra `Kconfig` files in external projects I think.

andrewboie, 2019-02-27:
Do we have PRs addressing all of these? High-priority bug, gating release.

ulfalizer, 2019-02-27:
PR that adds a CI check: https://github.com/zephyrproject-rtos/ci-tools/pull/21

ulfalizer, 2019-02-27:
@andrewboie 
IMO, not all of them need to be fixed before release (someone else set it to high priority).

Once the CI check is there, it's easy to kill off the stragglers, without new stuff getting introduced.

nashif, 2019-03-04:
reduced priority as most of the issue were fixed.

galak, 2019-03-07:
@ulfalizer did you open issues or fixes for these two:

```
CONFIG_DCD_DSR                      samples/net/wpan_serial/src/main.c:600

CONFIG_NET_DHCPV6                   samples/net/nats/src/main.c:135, samples/net/nats/src/main.c:174, samples/net/nats/src/main.c:269

```

galak, 2019-03-07:
Submitted PRs #14153 #14152 and #14151 to deal with last few Kconfig symbols that we don't have explicit issues for.

galak, 2019-03-07:
Closing this out as we have an issue open for the last cleanup related to this.

closed 2019-03-10 by galak