zephyrproject-rtos/zephyr #12150: Power management on nRF52 boards

Jiurifei, 2018-12-18:
I want to make my nRF52840 board work into low power mode.
So, in order to test the power consumption faster. I imitated the example: boards/nrf52/power_mgr, and used `nrf_power_task_trigger(NRF_POWER_TASK_LOWPWR);` to make device enter low power mode. But it didn't seem to work, the power dissipation was always the same. Then I took the  method of nordic SDK  into consideration, using `__WFE(); __SEV(); __WFE();` sequence, but it didn't work neither.
If anybody encounter this problem? Please give some suggestions.

Thank you
Jiurifei

JuricaResetar, 2018-12-18:
Hi.

I'm working on nrf52832 board and had no problem with sleep modes. I used the samples/subsys/power/power_mgr example and it works great!

The problem I have is understanding the code in details. For example, where in the documentation I can find details about sys_pm_notify_lps_entry function...? 

Thank you.

Jiurifei, 2018-12-18:
@JuricaResetar Thank you for your tip.  As far as I know, the location of `sys_pm_notify_lps_entry` is zephyr/subsys/power. And the whole function is  shown below:
`void sys_pm_notify_lps_entry(enum power_states state)
{
	printk("Entering Low Power state (%d)\n", state);
}
`
I don't understand what you want to know about this function, hope it can help.
What's more, have you tested the electricity of your board in different low power modes. Please let me konw, thank you!

JuricaResetar, 2018-12-18:
@Jiurifei I know what function does and when it's called. It obvious :D But my problems is, the function is not listed in the official documentation (I tried search also hehe). I worry how many functions are there which are not listed in the documentation... should I read all examples first...? Just rhetorical question :D
I've tested the power example on small custom made BLE modules, without any peripheral and the results are great. Around 0.2uA in deep sleep mode.

carlescufi, 2018-12-18:
@JuricaResetar thanks for helping out.
@Jiurifei are you satisfied with the response?

Jiurifei, 2018-12-18:
@JuricaResetar  Well, you know, open source project is usually like this :D. And as I said before, I have made my device get into deep sleep mode successfully. The power consumption was same as yours. But  what I care about is the electricity in other situation, like `SYS_PM_LOW_POWER_STATE`. Because, I want to let my board work in system on mode. I need to keep RTC running.So, I would appreciate it if you give more information.Also, I will try it tomorrow!

As I worked on example: samples/boards/nrf52/power_mgr. It is strange that once the device enter low power states except deep sleep state, then it would exit immediately. :o, I am a little confused.

@carlescufi  Thank you for concerning, I am satisfied and I would have a try.

Jiurifei, 2018-12-20:
If anybody has tested the power consumption in the state of  `SYS_POWER_STATE_CPU_LPS` or `SYS_POWER_STATE_CPU_LPS_1` ? Please let me know, thank you very much!

Jiurifei, 2018-12-27:
@JuricaResetar  Have you tested the power consumption in the state of `SYS_POWER_STATE_CPU_LPS `or `SYS_POWER_STATE_CPU_LPS_1` of you nrf52832 board? I have tried my best to make my board work at lower current, and I failed. The current now is about 300uA. I would appreciate it if you give me some advice.

JuricaResetar, 2019-01-03:
@Jiurifei A few days later and I'm facing the same error... My power consumption is around 270uA and does not go lower... :| It seems like the first few measurements were wrong... :( Did you make any progress?

tautologyclub, 2019-01-17:
Check the board defconfig for uarts and make sure it's disabled if you are not using uart. Make sure you don't have logging thread enabled. If using uart, enable device power management options and suspend uart0 before lps1 entry. If using uart1, then... Wait for a patch.

Jiurifei, 2019-01-19:
@tautologyclub These operations I have already token before. Logging thread was disabled all the time. And the current could only reduced around 20uA if I made uart  disabled. But  if I set the macro `config_serial=n` in the config file, then turned off the uart, it would save about 150uA power consumption. All in all, the current was still too high which was about 160uA finally. Same queston I  have raised in #12367.

tautologyclub, 2019-01-24:
The UART drains considerably more than 20uA so I doubt you managed to disable it successfully. Post your generated headers so we can verify - `autoconf.h` somewhere under `build/` in particular.

Jiurifei, 2019-01-25:
[autoconf.txt](https://github.com/zephyrproject-rtos/zephyr/files/2794433/autoconf.txt)
@tautologyclub the file I attached above is `autocong.h`, you can have a look, thank you.

sigvartmh, 2019-01-25:
@Jiurifei why is the board configure to `CONFIG_BOARD "nrf52_pca10040` ? this gives you `CONFIG_SOC "nRF52832_QFAA"` which is not the same SoC as stated above `nRF52840`

tautologyclub, 2019-01-25:
Hm. Perhaps it's board specific. I've only tried it on 52840 and it draws 4uA in LPS1. Is the debugger connected? That can make a huge difference

Jiurifei, 2019-01-25:
@tautologyclub Can you tell me how you achieved that and which sample you used? As I am sure there was no debugger connected when I tested the power consumption. If the debugger was connected, the current would rise more than 2mA rather than hundreds of uA.

tautologyclub, 2019-01-28:
I tried both the nrf power management sample and my own code. In both cases, the key was ensuring UART was disabled (either through the device_pm API or by setting appropriate CONFIGs to `n`), and ensuring `sys_set_power_state(SYS_POWER_STATE_CPU_LPS_1)` was actually called.

Jiurifei, 2019-01-29:
@tautologyclub thank you for suggestions. In fact,  I have tried several methods to reduce power consumption of the board, including disabling UART and turning off some other relevant macros in config file. The result is that the current would reduce 150uA when UART disabled, but it was still unacceptable. I doubt that other factors caused this to happen. So could you please post `.config` file? I may get some important information from that.

closed 2020-10-04 by github-actions[bot]