zephyrproject-rtos/zephyr #12154: 802.15.4 6LowPAN stopped working with multiple samples

rlubos, 2018-12-18:
**Describe the bug**
6LowPAN over 802.15.4 stopped working recently in multiple samples. A quick bisect showed up that the regression was introduced with #11374. Bisecting stopped specifically at commit https://github.com/zephyrproject-rtos/zephyr/commit/66cc034a34432fd3df462fa1fe23627b10948891.

For instance in echo_server, during initialization sample throws a bunch of errors:
```
[00:00:00.000,000] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.005,706] <inf> net_config: Initializing network
[00:00:00.011,962] <err> net_ieee802154_frame: Could not generate data frame 15 vs 13
[00:00:00.011,993] <err> net_ieee802154_frame: Could not generate data frame 15 vs 13
[00:00:00.015,441] <err> net_ieee802154_frame: Could not generate data frame 15 vs 13
[00:00:00.109,863] <inf> net_config: IPv6 address: fe80::140f:2272:a690:223d
[00:00:00.109,954] <inf> net_config: IPv6 address: fe80::140f:2272:a690:223d
[00:00:01.008,087] <err> net_ieee802154_frame: Could not generate data frame 15 vs 13
[00:00:02.015,991] <err> net_ieee802154_frame: Could not generate data frame 15 vs 13
```

I've observed this behavior with echo_client/echo_server samples, but I've seen it was reported on slack as well in RPL sample.

Adding some logs shows that `ieee802154_compute_header_size` function is provided with some weird adresses for computation, for instance:
```
[00:00:00.039,794] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (0x20001264): Computing size for 14:f22:72a6:9022:3d00:1:ff90:223d
[00:00:00.049,713] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (0x20001264): Computed size of 13
[00:00:00.060,455] <dbg> net_ieee802154_frame.get_dst_addr_mode: (0x20001264): Broadcast destination
[00:00:00.072,906] <err> net_ieee802154_frame: Could not generate data frame 15 vs 13
```
```
[00:00:00.205,657] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (0x20001264): Computing size for ::100:1:ff00:1
[00:00:00.216,979] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (0x20001264): Computed size of 13
[00:00:00.228,454] <dbg> net_ieee802154_frame.get_dst_addr_mode: (0x20001264): Broadcast destination
[00:00:00.238,586] <err> net_ieee802154_frame: Could not generate data frame 15 vs 13
```

**To Reproduce**
Just run echo_client/server sample with overlay-802154.conf. I've tested on nrf52840_pca10056 board.

**Expected behavior**
802.15.4 L2 does not throw errors during frame creation.

**Impact**
Cannot effectively test any changes related to 802.15.4 with curent master

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain -Zephyr SDK
 - Commit - current master

rlubos, 2018-12-18:
@tbursztyka FYI

tbursztyka, 2018-12-18:
Will take a look

tbursztyka, 2019-01-07:
Ok I could take some time to verify this. Looks like something is fishy in ipv6. I made kind of same logging modif as yours: See my logs:

```
[00:00:00.444,247] <inf> net_config: IPv6 address: fe80::210:2030:35a9:6324
[00:00:00.482,437] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (0x00401018): Address 2:1020:3035:a963:2400:1:ffa9:6324
[00:00:00.482,876] <dbg> net_ieee802154_frame.ieee802154_compute_header_size: (0x00401018): Computed size of 13
[00:00:00.483,033] <dbg> net_ieee802154_frame.get_dst_addr_mode: (0x00401018): Broadcast destination
```

Address should be the one shown earlier from net_config, and it's a mix of it and garbage. I'll investigate that.

PR #12370 (2019-01-08): net/ieee802154: Compute ll hdr size before 6lo compression
Or then the IP header access will be bogus afterwards.

Fixes #12154
commit dc9e408e (2019-01-08): net/ieee802154: Compute ll hdr size before 6lo compression

Or then the IP header access will be bogus afterwards.

Fixes #12154
closed 2019-01-08 by jukkar