zephyrproject-rtos/zephyr #12160: settings: NVS back-end

nvlsianpu, 2018-12-18:
NVS was designed as ultra simple FS which is supposed to store configuration values very effectively.
It will be nice to have back-end for it - it can be used instead of fcb back-end. I expect similar complexity, but much  better performance as NVS is traversable in both direction and It is able to take care about entry overriding by the implementation.

Laczen, 2019-01-03:
@nvlsianpu, implementing a nvs backend for the settings module is not straight forward. It is mainly limited by the fact that settings supposes that data is stored in the backend as a line consisting of "name"=value. This is of course not good for flash usage as each name is stored in flash again when only the value is changing.
Removing this supposition requires quite some work. To allow for easy extension of the settings module I am proposing the following changes:
a. Change the api for set: value_ctx becomes a structure like:
```
value_ctx {
   size_t len;
   size_t (*read)(char *data, size_t len, void *arg);
   void *arg;
}
```
In the set routine value_ctx.len and value_ctx.read(&data, len, arg) can be used and arg contains information that is needed by the backend.
b. The save routine checks for duplicate data in the backend
c. The delete routine should also delete all items that have a name starting with the deleted entry name.

nvlsianpu, 2019-01-07:
Generally the idea sound good.

b - this mean that you want to move comparation rutine to backends?
c - this sounds like a new feature (which I like).

Laczen, 2019-03-25:
I will no longer work on trying to add a nvs backend for the settings subsystem, so for me this can be closed.

nvlsianpu, 2019-06-26:
This is the early bird of NVS backed By @kapi-no  : https://github.com/kapi-no/zephyr/commits/settings_nvs_backend

PR #17124 (2019-07-10): Settings with NVS backend
Added support for NVS backend in Settings with following functionalities:

- full tree loading with settings_load()
- subtree loading with settings_load_subtree()
- subtree / single entry deletion with settings_delete()
- single entry update / write with settings_save_one()

Fixes #12160
closed 2019-07-10 by ioannisg