zephyrproject-rtos/zephyr #12162: net: icmpv4: Zephyr replies to ICMPv4 packets with broadcast destination IPv4 address

rmstoi, 2018-12-18:
Zephyr replies to ICMPv4 packet with destination IP address set to 255.255.255.255 broadcast address.

These packets should be dropped instead.

[icmpv4-bcast-dest.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2690655/icmpv4-bcast-dest.pcap.gz)

tbursztyka, 2019-01-03:
Can you detail a bit which icmpv4 packet type? (echo req?)

Note there is an option: CONFIG_NET_ICMPV4_ACCEPT_BROADCAST which is set by default (I think it should not actually). Unset it and re-run the test.

rmstoi, 2019-01-04:
The packet is an echo request. If config CONFIG_NET_ICMPV4_ACCEPT_BROADCAST is unset the test case passes. RFC 1122 in 3.2.2.6 says "ICMP Echo Request destined to an IP broadcast or IP multicast address MAY be silently discarded." Config should probably be disabled by default.

tbursztyka, 2019-01-04:
Thanks for testing this, let's disable it by default

PR #12371 (2019-01-08): net/icmpv4: Disable broadcast echo request reply by default
Accepting broadcast echo request and replying to it could provide an
attack vector.

Fixes #12162
commit b04e7190 (2019-01-08): net/icmpv4: Disable broadcast echo request reply by default

Accepting broadcast echo request and replying to it could provide an
attack vector.

Fixes #12162
closed 2019-01-08 by jukkar