zephyrproject-rtos/zephyr #12170: logging: previous changes to dropped message notification break wifi sample debug

vanti, 2018-12-19:
**Describe the bug**
The wifi sample can no longer be debugged in gdb with CONFIG_XIP=n on CC3220 after the changes in commit id a211c42a2976592aa57e8076997cc5e81963f3c6, according to 'git bisect'. The example prints out the uart prompt, then hits a fault (__usage_fault is executed).

Specifically, adding the else clause in msg_process() from subsys/logging/log_core.c strangely introduces the failure:

```
       else {
                atomic_inc(&dropped_cnt);
        }
```

Workaround is to comment out this else clause (and probably break logging in the process).
Program seems to run fine when flashed (CONFIG_XIP=y).

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake .. -DBOARD=cc3220_launchxl
3. make
4. arm-zephyr-eabi-gdb -x ~/ti/gdbinit_cc3220 zephyr/zephyr.elf
5. In gdb, type 'c' to run

_Contents in gdbinit_cc3220:_
```
set remotetimeout 10000
target extended-remote | openocd -c "gdb_port pipe; log_output openocd.log" -f /usr/local/share/openocd/scripts/board/ti_cc3220sf_launchpad.cfg
monitor soft_reset_halt
load
```

**Expected behavior**
The wifi sample should run in gdb in the same way as flashed.

**Impact**
Inconvenience when trying to debug

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK

pfalcon, 2018-12-27:
cc @nordic-krch

GAnthony, 2019-01-10:
@nordic-krch, any ideas on why this change might be causing a __usage_fault?

nordic-krch, 2019-01-10:
@GAnthony that sounds strange. Do you have stack guards enabled to check if maybe stack for some thread is too small?

GAnthony, 2019-01-11:
@nordic-krch, thanks for the response.
Given the program runs OK from flash vs SRAM, seems it may not be a stack problem, but perhaps more a timing issue.
But before going too deep into the debug path, we're just curious if any thoughts how this could affecting timing, maybe introducing a race somewhere?

From the commit message:

> It is optional for a backend to implement handler.

Is there a CONFIG option for us to disable it?

nordic-krch, 2019-03-13:
@vanti are you able to still reproduce the issue? How does it crash? Any output printed on uart?

vanti, 2019-03-13:
I can no longer reproduce the issue with CONFIG_XIP=n. UART output is now correctly printed out on as of c0183fdeddf48479d5698da7f8f9fa8bf6f540d4. Wifh CONFIG_XIP=y, we understand there is currently an issue in openocd (with how it is resetting the board during load) that may have caused the issue. So closing this issue for now.

closed 2019-03-13 by vanti