zephyrproject-rtos/zephyr #12171: ot: ping ten times will crash

turon, 2018-12-19:
**Description**
 I observed that a device initiating a ping of 10x 100 bytes will crash:

```ot cmd ping ff02::1 100 10```

**Build**
Steps to reproduce the behavior:
1. cd samples/net/echo_client
2. mkdir -p build/nrf52
3. cd build/nrf52
4. cmake -GNinja -DCONF_FILE="prj.conf overlay-ot.conf" -DBOARD=nrf52840_pca10056 ../..
5. ninja
6. ninja flash
7. sudo minicom -D /dev/ttyACM0 -b 115200
8. ot cmd panid 0xabcd 
9. ot cmd ifconfig up
10. ot cmd thread start
11. Setup a second node, flash, and run steps 8-10
12. ot cmd state --> confirm Leader on one node and Child or Router on the other
12. ot cmd ping ff02::1
13. Observe response
14. ot cmd ping ff02::1 100 10
15. Observe crash

**Expected behavior**
Expect to see 10 ping responses and shell app to continue to run.

rlubos, 2019-01-21:
@turon I've identified the issue, apparently it's an OpenThread bug - it does not handle ping timer correctly for `cli_console` backend. I've posted a fix to the OpenThread:
https://github.com/openthread/openthread/pull/3486

Once accepted and merged, we can update the OpenThread in Zephyr.

jarz-nordic, 2019-01-30:
@turon  It seems to be fixed on shell side with PR #12747.

PR #12869 (2019-01-31): net: openthread: Bump OpenThread commit to f9d757a1
This version has cli_console backend fixed, hence no longer crashes on `ot ping` command.

Fixes #12171
closed 2019-01-31 by nashif