zephyrproject-rtos/zephyr #12186: net: arp: Zephyr does use MAC address from unicast ARP reply with wrong target MAC

rmstoi, 2018-12-19:
If Zephyr receives unrequested unicast ARP reply with wrong target MAC it does not update translation table. So, instead of using destination MAC address from request packet (ICMP, TCP, UDP) in response, it should use MAC from unrequested ARP reply.

According to RFC 5227: 

> An incorrect ARP implementation may ignore ARP Replies where 'ar$tha' doesn't match its hardware 
> address.

[unreq-arp-reply-wrong-tgt-mac.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2695533/unreq-arp-reply-wrong-tgt-mac.pcap.gz)

tbursztyka, 2019-01-03:
So Zephyr drops unrequested unicast ARP reply and that's a bad thing? Ok, sounds weird.

Could you explain "So, instead of using destination MAC address from request packet (ICMP, TCP, UDP) in response, it should use MAC from unrequested ARP reply."
On arp request we don't know the MAC address, so there is none to use at this point (thus the ARP request). Do you mean we should blindly update the table from the info read on the unrequested ARP reply and use such info for pending packets? (ones that generated an ARP request on our side)

rmstoi, 2019-01-04:
> Do you mean we should blindly update the table from the info read on the unrequested ARP reply and use such info for pending packets? (ones that generated an ARP request on our side)

Yes, that is what I mean.

jukkar, 2019-01-16:
We do not support unsolicited ARP replies and IMHO that is a good thing from security point of view.

closed 2019-01-16 by jukkar