zephyrproject-rtos/zephyr #12190: Mbedtls MBEDTLS_PLATFORM_STD_SNPRINTF issue

karthikprabhu17, 2018-12-19:
**Describe the bug**
I just pull latest from master and this happened:
Trying to build MQTT app which uses mbedtls  and I keep getting missing Mbedtls functions

**/Users/kprabhuv/new_zephyr/ext/lib/crypto/mbedtls/library/platform.c:123:64: error: expected expression before ';' token
                           ... ) = MBEDTLS_PLATFORM_STD_SNPRINTF;
                                                                ^
make[2]: *** [zephyr/ext/lib/crypto/mbedtls/CMakeFiles/ext__lib__crypto__mbedtls.dir/library/platform.c.obj] Error 1
make[1]: *** [zephyr/ext/lib/crypto/mbedtls/CMakeFiles/ext__lib__crypto__mbedtls.dir/all] Error 2
make: *** [all] Error 2**

**To Reproduce**
Steps to reproduce the behavior:
Unable to reproduce this using sample apps i know.

**Expected behavior**
Clear build

**Impact**
showstopper. Unable to build.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain : gnuarmemb
 - Commit: 3f422e2efe25b67f62854df5721fd51217b9d725

**Additional context**
Add any other context about the problem here.

karthikprabhu17, 2018-12-19:
Just verified this happens only on the MAC and not on an Ubuntu machine!

SebastianBoe, 2018-12-28:
Hi,

what does this mean?

"Unable to reproduce this using sample apps i know."

Please try to reproduce the issue using a minimal patch to an existing zephyr sample or test.

You could also try to find out what MBEDTLS_PLATFORM_STD_SNPRINTF is being defined to be, that might give a clue to what is going wrong. This could be done by building verbosely and then copy-editting the command that files with the flag "-E" (this will run the preprocessor only, allowing one to inspect the MBEDTLS_PLATFORM_STD_SNPRINTF symbol).

galak, 2019-02-01:
which gnuarmemb version are you using?

karthikprabhu17, 2019-02-04:
@galak sorry for the delay in response. I am using gcc-arm-none-eabi-7-2018-q2-update. I was able to get rid of the error by removing MBEDTLS_PLATFORM_STD_SNPRINTF from config file. Although that is not solution i know!

galak, 2019-02-22:
Sorry for taking a while to respond, can you let me know how to reproduce?

karthikprabhu17, 2019-03-01:
I just tried it...the master dos not have this problem . Closing this.

closed 2019-03-01 by karthikprabhu17