zephyrproject-rtos/zephyr #12192: include: __assert: enabling assert causes build errors

vanti, 2018-12-19:
**Describe the bug**
When building the http_get sample for cc3220sf_launchxl, errors were observed:
    In file included from /home/vwan/zephyr/include/kernel_includes.h:22:0,
                     from /home/vwan/zephyr/include/kernel.h:17,
                     from /home/vwan/zephyr/include/logging/log_msg.h:9,
                     from /home/vwan/zephyr/include/logging/log_core.h:9,
                     from /home/vwan/zephyr/include/logging/log.h:11,
                     from /home/vwan/zephyr/drivers/wifi/simplelink/simplelink_log.h:17,
                     from /home/vwan/zephyr/drivers/wifi/simplelink/simplelink_support.c:11:
    /home/vwan/zephyr/drivers/wifi/simplelink/simplelink_support.c: In function ‘configure_simplelink’:
    /home/vwan/zephyr/include/misc/__assert.h:113:11: error: called object is not a function or function pointer
        printk("\t" fmt "\n", ##__VA_ARGS__);      \
               ^
    /home/vwan/zephyr/drivers/wifi/simplelink/simplelink_support.c:29:33: note: in expansion of macro ‘__ASSERT’
     #define ASSERT_ON_ERROR(ret, e) __ASSERT(ret >= 0, e)
                                     ^~~~~~~~
    /home/vwan/zephyr/drivers/wifi/simplelink/simplelink_support.c:83:2: note: in expansion of macro ‘ASSERT_ON_ERROR’
      ASSERT_ON_ERROR(mode, DEVICE_ERROR);
      ^~~~~~~~~~~~~~~
    /home/vwan/zephyr/include/misc/__assert.h:113:20: error: expected ‘)’ before string constant
        printk("\t" fmt "\n", ##__VA_ARGS__);      \

This change, introduced in commit 6825d91a1b459bfd9b658603514e9a120264e7de, seems to be causing the error:
+                       printk("\t" fmt "\n", ##__VA_ARGS__);      \

Removing "\t" and "\n" allows the build to proceed.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/net/sockets/http_get
2. mkdir build; cd build
3. cmake -DBOARD=cc3220sf_launchxl
4. make
5. See error

**Expected behavior**
I expect the sample to build with CONFIG_ASSERT=y.

**Impact**
This prevents assertions from being enabled, and is an annoyance when debugging.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.9.5
 - Commit SHA where this was reproduced: 0b9cfe70ed261c88a111970e4f36c748c748b8ff

GAnthony, 2018-12-19:
FYI: @kumar, there is a sample.yaml in this http_get sample directory, which I would have expected to catch this build error.   Is it not correct?
Are the Zephyr build tests building *anything* for cc3220sf_launchxl?

vanti, 2018-12-21:
The PR would not prevent others from running into the same issue, but it would allow http_get and other samples using the simplelink driver to build with assertions enabled.

PR #12217 (2018-12-26): drivers: wifi: simplelink: Fix strings used in assertions
Samples for cc3220sf_launchxl where assertions are enabled
(e.g. http_get) fail to build because some of the strings used in the
simplelink wifi driver have parentheses around them. This breaks the
current implementation of the __ASSERT macro. This commit removes
the parentheses, which are unnecessary.

http_get has been verified to build after this change.

Fixes #12192
commit 0f64ec5c (2018-12-26): drivers: wifi: simplelink: Fix strings used in assertions

Samples for cc3220sf_launchxl where assertions are enabled
(e.g. http_get) fail to build because some of the strings used in the
simplelink wifi driver have parentheses around them. This breaks the
current implementation of the __ASSERT macro. This commit removes
the parentheses, which are unnecessary.

http_get has been verified to build after this change.

Fixes #12192
closed 2018-12-26 by nashif