zephyrproject-rtos/zephyr #12207: Repeated code?

zzzh, 2018-12-21:
In [https://github.com/zephyrproject-rtos/zephyr/blob/master/kernel/device.c](url)
Line 84 to 92 seems to be a repeat of line 77 to 82?

`struct device *_impl_device_get_binding(const char *name)
{
	struct device *info;

	/* Split the search into two loops: in the common scenario, where
	 * device names are stored in ROM (and are referenced by the user
	 * with CONFIG_* macros), only cheap pointer comparisons will be
	 * performed.  Reserve string comparisons for a fallback.
	 */
	for (info = __device_init_start; info != __device_init_end; info++) {
		if ((info->driver_api != NULL) &&
		    (info->config->name == name)) {
			return info;
		}
	}

	for (info = __device_init_start; info != __device_init_end; info++) {
		if (info->driver_api == NULL) {
			continue;
		}

		if (strcmp(name, info->config->name) == 0) {
			return info;
		}
	}

	return NULL;
}`

andrewboie, 2018-12-21:
It's not, the first pass does a direct pointer comparison and the second pass uses strcmp.
The comment (which you quoted) explains this..

zzzh, 2018-12-24:
The 2 ways looks same to me.

zzzh, 2018-12-24:
I think they looks same to most people.

closed 2019-04-01 by andrewboie