zephyrproject-rtos/zephyr #12215: Nordic: remove default-enable of bluetooth capability

pabigot, 2018-12-21:
As discussed in #12209 it's been discovered that (almost?) all Nordic-based boards have the following in their board defconfig:

```
# bluetooth
CONFIG_BT=y
CONFIG_BT_CTLR=y
```

This causes any application built for the boards to get roughly 30 KB of possibly unnecessary driver infrastructure.  It's particularly an issue when another transport mechanism (802.15.4 or OpenThreads) is desired instead.

Task is to review why this was done and, if there isn't a compelling explanation, remove this from all Nordic board defconfig files.

carlescufi, 2018-12-21:
I agree with getting this done, I will send a PR. Changed from bug to enhancement.

SebastianBoe, 2018-12-27:
This is a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/5454

I agree that boards should not be enabling large software stacks, like BT.

pabigot, 2019-01-13:
It would be good to get a fix in place before topic-counter goes in.  Default-enabling bluetooth causes any attempt to use `COUNTER_TIMER0` or `COUNTER_RTC0` to be rejected by cmake with an unhelpful diagnostic.  It took a while to figure out the cause was `CONFIG_BT=y`.

carlescufi, 2019-02-07:
@cvinayak @SebastianBoe how about we do it like the particle boards do:
https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/particle_xenon/Kconfig.defconfig#L46

That way we only enable the controller if Bluetooth is enabled.

PR #13142 (2019-02-07): boards: arm: nrf: Make Bluetooth optional
Instead of enabling Bluetooth by default on nRF5x boards, only enable
the controller if Bluetooth has been enabled by the applicaiton.

Fixes #5454
Fixes #12215
commit 9f1acfe2 (2019-02-07): boards: arm: nrf: Make Bluetooth optional

Instead of enabling Bluetooth by default on nRF5x boards, only enable
the controller if Bluetooth has been enabled by the applicaiton.

Fixes #5454
Fixes #12215
closed 2019-02-07 by nashif