zephyrproject-rtos/zephyr #12224: POSIX api are incompatible with arm gcc 2018q2 toolchain

madandylau, 2018-12-24:
**Describe the bug**
I made a simple pthread demo based on the hello_world sample.(The current git version of Zephyr is used) Below is the main.c source code:

```
#include <zephyr.h>
#include <posix/pthread.h>

#include <stdio.h>

#define MY_STACK_SIZE 1024
K_THREAD_STACK_DEFINE(my_stack_area, MY_STACK_SIZE);

void *my_thread(void* parm)
{
	int i = 0;

	while(1) {
		printf("my_thread count = %d\n", i++);
		k_sleep(250);
	}
}

void main(void)
{
	pthread_t my_thread_id;
	pthread_attr_t my_thread_attr;
	int i = 0;

	pthread_attr_init(&my_thread_attr); 
	my_thread_attr.stack = my_stack_area;
	my_thread_attr.stacksize = MY_STACK_SIZE;

	pthread_attr_setdetachstate(&my_thread_attr, PTHREAD_CREATE_DETACHED);
	pthread_create(&my_thread_id, &my_thread_attr, my_thread, NULL);
	pthread_join(my_thread_id, NULL);

	while(1) {
		printf("main count = %d\n", i++);
		k_sleep(1000);
	}
}
```
**Compiling with zephyr sdk 0.9.5**
The code works very well when compiling with zephyr sdk 0.9.5. Here is the detailed step:
1. source zephyr-env.sh
export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_SDK_INSTALL_DIR="my_sdk_path"
2. cd samples/hello_world && mkdir build && cd build
3. cmake -GNinja -DBOARD=qemu_cortex_m3 ..
4. ninja menuconfig
Check "C Library ---> Build with newlib c library"
Check "C Library ---> Send stdout to console"
Check "Additional libraries ---> POSIX APIs"
Check "Additional libraries ---> POSIX PTHREAD IPC API"
5. ninja && ninja run

**Compiling with gcc-arm-none-eabi-7-2018-q2-update**
Steps are almost the same except step 1:
1. source zephyr-env.sh
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
export GNUARMEMB_TOOLCHAIN_PATH="my_arm_toolchain_path"
export ZEPHYR_SDK_INSTALL_DIR="my_sdk_path"

When compiling, there comes a lot of conflicting type problems. I try to fix it by myself. But after hours trying. I find that the incompatibility is too complicated for me to handle.

I think the main reason is that the arm gcc 2018q2 toolchain use newlib of version 3.0.0. But zephyr sdk use the version 2.4.0.

Do you guys have any idea what should I do? Thanks and merry christmas : )

**Environment**
 - OS: Ubuntu 18.04 (the cmake version is lower than required, so I need to use the cmake contained in the sdk)

SebastianBoe, 2018-12-27:
@aescolar : FYI

pfalcon, 2018-12-27:
@madandylau: Thanks for the detailed bug report! But you seem to missed the most important part.

> When compiling, there comes a lot of conflicting type problems.

Which conflicting type problems exactly? Please quote the error messages.

pfalcon, 2018-12-27:
And you report issues with arm gcc. Any reason you don't use Zephyr SDK's toolchain?

madandylau, 2018-12-28:
@pfalcon Yes, the toolchain I used is from [here](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads). 

First, I need to compile the Swift runtime(which is written in C++) with Clang, and link them with libstdc++ which provided by the arm gcc toolchain. This is the main reason.

Secondly, I want to create an Arduino like IDE but for Swift language later. Seems the arm gcc toolchain has a better support for Windows and Mac OS environment right now. That's very important for my project. 

Here is the diagnoses:
```
liu@ubuntu:~/zephyr-master/samples/hello_world/build$ ninja
[0/1] Re-running CMake...
-- Selected BOARD qemu_cortex_m3
Zephyr version: 1.13.99
Parsing Kconfig tree in /home/liu/zephyr-master/Kconfig
Loading /home/liu/zephyr-master/samples/hello_world/build/zephyr/.config as base
Configuration written to '/home/liu/zephyr-master/samples/hello_world/build/zephyr/.config'
-- git describe failed: fatal: not a git repository (or any of the parent directories): .git; "1.13.99" will be used instead
-- Loading /home/liu/zephyr-master/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts as base
-- Overlaying /home/liu/zephyr-master/dts/common/common.dts
-- Cache files will be written to: /home/liu/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/liu/zephyr-master/samples/hello_world/build
[1/103] Preparing syscall dependency handling

[45/103] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/liu/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/liu/zephyr-master/kernel/include -I/home/liu/zephyr-master/arch/arm/include -I/home/liu/zephyr-master/soc/arm/ti_lm3s6965 -I/home/liu/zephyr-master/soc/arm/ti_lm3s6965/include -I/home/liu/zephyr-master/soc/arm/include -I/home/liu/zephyr-master/include -I/home/liu/zephyr-master/include/drivers -Izephyr/include/generated -isystem /home/liu/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/include -isystem /home/liu/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/include-fixed -I/home/liu/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include -I/home/liu/zephyr-master/ext/hal/cmsis/Include -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/liu/zephyr-master/samples/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-common -mthumb -mcpu=cortex-m3 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7-m -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c ../src/main.c
In file included from /home/liu/zephyr-master/include/posix/pthread.h:13:0,
                 from ../src/main.c:2:
/home/liu/zephyr-master/include/posix/unistd.h:19:22: error: conflicting types for 'mode_t'
 typedef unsigned int mode_t;
                      ^~~~~~
In file included from /home/liu/zephyr-master/include/posix/time.h:30:0,
                 from /home/liu/zephyr-master/include/posix/pthread.h:12,
                 from ../src/main.c:2:
/home/liu/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/sys/types.h:205:18: note: previous declaration of 'mode_t' was here
 typedef __mode_t mode_t;  /* permissions */
                  ^~~~~~
[54/103] Building C object zephyr/lib/libc/newlib/CMakeFiles/lib__libc__newlib.dir/libc-hooks.c.obj
ninja: build stopped: subcommand failed.
```

I try to fix this myself, but it just comes out more and more errors. 

During my debugging, I find the newlib 2.4.0 and 3.0.0 have some tiny differences. I don't know if this a big problem.

For example, here is the disassembly of _close_r from the sdk's libc.a:
```
lib_a-closer.o:     file format elf32-littlearm

Disassembly of section .text:

00000000 <_close_r>:
   0:	e3a03000 	mov	r3, #0
   4:	e92d4070 	push	{r4, r5, r6, lr}
   8:	e59f4028 	ldr	r4, [pc, #40]	; 38 <_close_r+0x38>
   c:	e1a05000 	mov	r5, r0
  10:	e1a00001 	mov	r0, r1
  14:	e5843000 	str	r3, [r4]
  18:	ebfffffe 	bl	0 <close>
  1c:	e3700001 	cmn	r0, #1
  20:	1a000002 	bne	30 <_close_r+0x30>
  24:	e5943000 	ldr	r3, [r4]
  28:	e3530000 	cmp	r3, #0
  2c:	15853000 	strne	r3, [r5]
  30:	e8bd4070 	pop	{r4, r5, r6, lr}
  34:	e12fff1e 	bx	lr
  38:	00000000 	andeq	r0, r0, r0
```
It jumps to "close" function

And here is the disassembly of _close_r from the arm gcc toolchain:
```
./lib_a-closer.o:     file format elf32-littlearm

Disassembly of section .text._close_r:

00000000 <_close_r>:
   0:	b538      	push	{r3, r4, r5, lr}
   2:	4c07      	ldr	r4, [pc, #28]	; (20 <_close_r+0x20>)
   4:	2300      	movs	r3, #0
   6:	4605      	mov	r5, r0
   8:	4608      	mov	r0, r1
   a:	6023      	str	r3, [r4, #0]
   c:	f7ff fffe 	bl	0 <_close>
  10:	1c43      	adds	r3, r0, #1
  12:	d000      	beq.n	16 <_close_r+0x16>
  14:	bd38      	pop	{r3, r4, r5, pc}
  16:	6823      	ldr	r3, [r4, #0]
  18:	2b00      	cmp	r3, #0
  1a:	d0fb      	beq.n	14 <_close_r+0x14>
  1c:	602b      	str	r3, [r5, #0]
  1e:	bd38      	pop	{r3, r4, r5, pc}
  20:	00000000 	andeq	r0, r0, r0

```
It jumps to "_close" function.

ramakrishnapallala, 2019-01-18:
I am tot an expert in compiler related stuff. Please re-assign.

nslowell, 2019-01-21:
@youvedeep: FYI

nslowell, 2019-02-01:
@madandylau Can you apply this patch.  It will likely move you to the next error.  When you get that error, can you copy the gcc command that failed and add "-H" to the options and provide the output?
[0001-posix-define-mode_t-in-sys-types.h-only-if-it-isn-t-.txt](https://github.com/zephyrproject-rtos/zephyr/files/2822145/0001-posix-define-mode_t-in-sys-types.h-only-if-it-isn-t-.txt)

pfalcon, 2019-02-01:
Another ticket related to POSIX API issues with specific action items: https://github.com/zephyrproject-rtos/zephyr/issues/12965 . (No claims it would help here, just cross-linking related issues.)

pfalcon, 2019-02-01:
>  galak added the priority: medium label 24 days ago

@galak, regarding this, I think that we should concentrate on making POSIX subsys to work well with Zephyr's own SDK first (and there're enough issues there). So, I'd suggest to that this as prio: low.

madandylau, 2019-02-12:
@nslowell , sorry for this late reply. I download the new 1.14.0-rc1 version and did the test you mentioned.

When I run ninja, the output are as follows:

```
liu@ubuntu:~/zephyr-1.14.0-rc1_pthread/samples/hello_world/build$ ninja
[0/1] Re-running CMake...
Zephyr version: 1.14.0
-- git describe failed: fatal: not a git repository (or any of the parent directories): .git; "1.14.0-rc1" will be used instead
-- Selected BOARD qemu_cortex_m3
-- Loading /home/liu/zephyr-1.14.0-rc1_pthread/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts as base
-- Overlaying /home/liu/zephyr-1.14.0-rc1_pthread/dts/common/common.dts
Parsing Kconfig tree in /home/liu/zephyr-1.14.0-rc1_pthread/Kconfig
Loading /home/liu/zephyr-1.14.0-rc1_pthread/samples/hello_world/build/zephyr/.config as base
Configuration written to '/home/liu/zephyr-1.14.0-rc1_pthread/samples/hello_world/build/zephyr/.config'
-- Cache files will be written to: /home/liu/.cache/zephyr
Including module(s):
-- Configuring done
-- Generating done
-- Build files have been written to: /home/liu/zephyr-1.14.0-rc1_pthread/samples/hello_world/build
[1/104] Preparing syscall dependency handling

[61/104] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/sleep.c.obj
FAILED: zephyr/lib/posix/CMakeFiles/lib__posix.dir/sleep.c.obj
ccache /home/liu/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I../../../kernel/include -I../../../arch/arm/include -I../../../include -I../../../include/drivers -Izephyr/include/generated -I../../../soc/arm/ti_lm3s6965 -I/home/liu/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include -I../../../ext/hal/cmsis/Include -I../../../include/posix -isystem /home/liu/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/include -isystem /home/liu/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/include-fixed -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/liu/zephyr-1.14.0-rc1_pthread/samples/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-common -mthumb -mcpu=cortex-m3 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -fno-short-enums -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7-m -std=c99 -MD -MT zephyr/lib/posix/CMakeFiles/lib__posix.dir/sleep.c.obj -MF zephyr/lib/posix/CMakeFiles/lib__posix.dir/sleep.c.obj.d -o zephyr/lib/posix/CMakeFiles/lib__posix.dir/sleep.c.obj   -c /home/liu/zephyr-1.14.0-rc1_pthread/lib/posix/sleep.c
In file included from ../../../include/posix/unistd.h:17:0,
                 from /home/liu/zephyr-1.14.0-rc1_pthread/lib/posix/sleep.c:8:
../../../include/fs.h:138:2: error: expected specifier-qualifier-list before 'ssize_t'
  ssize_t (*read)(struct fs_file_t *filp, void *dest, size_t nbytes);
  ^~~~~~~
../../../include/fs.h:247:1: error: unknown type name 'ssize_t'; did you mean '_ssize_t'?
ssize_t fs_read(struct fs_file_t *zfp, void *ptr, size_t size);
^~~~~~~
_ssize_t
../../../include/fs.h:268:1: error: unknown type name 'ssize_t'; did you mean '_ssize_t'?
ssize_t fs_write(struct fs_file_t *zfp, const void *ptr, size_t size);
^~~~~~~
_ssize_t
../../../include/fs.h:286:36: error: unknown type name 'off_t'; did you mean '_off_t'?
int fs_seek(struct fs_file_t *zfp, off_t offset, int whence);
                                    ^~~~~
                                    _off_t
../../../include/fs.h:298:1: error: unknown type name 'off_t'; did you mean '_off_t'?
off_t fs_tell(struct fs_file_t *zfp);
^~~~~
_off_t
../../../include/fs.h:318:40: error: unknown type name 'off_t'; did you mean '_off_t'?
int fs_truncate(struct fs_file_t *zfp, off_t length);
                                        ^~~~~
                                        _off_t
In file included from /home/liu/zephyr-1.14.0-rc1_pthread/lib/posix/sleep.c:8:0:
../../../include/posix/unistd.h:24:8: error: unknown type name 'ssize_t'
extern ssize_t write(int file, const void *buffer, size_t count);
        ^~~~~~~
../../../include/posix/unistd.h:25:8: error: unknown type name 'ssize_t'
extern ssize_t read(int file, void *buffer, size_t count);
        ^~~~~~~
../../../include/posix/unistd.h:26:8: error: unknown type name 'off_t'
extern off_t lseek(int file, off_t offset, int whence);
        ^~~~~
../../../include/posix/unistd.h:26:30: error: unknown type name 'off_t'; did you mean '_off_t'?
extern off_t lseek(int file, off_t offset, int whence);
                              ^~~~~
                              _off_t
[70/104] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread.c.obj
```

When I add "-H" to the failed command, the output are as follows:

```
[... 119 lines cut ...]
../../../include/fs.h:318:40: error: unknown type name 'off_t'; did you mean '_off_t'?
int fs_truncate(struct fs_file_t *zfp, off_t length);
                                        ^~~~~
                                        _off_t
In file included from /home/liu/zephyr-1.14.0-rc1_pthread/lib/posix/sleep.c:8:0:
../../../include/posix/unistd.h:24:8: error: unknown type name 'ssize_t'
extern ssize_t write(int file, const void *buffer, size_t count);
        ^~~~~~~
../../../include/posix/unistd.h:25:8: error: unknown type name 'ssize_t'
extern ssize_t read(int file, void *buffer, size_t count);
        ^~~~~~~
[... cut ...]
extern off_t lseek(int file, off_t offset, int whence);
                              ^~~~~
                              _off_t
Multiple include guards may be useful for:
/home/liu/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/limits.h
/home/liu/gcc-arm-none-eabi-7-2018-q2-update/lib/gcc/arm-none-eabi/7.3.1/include-fixed/syslimits.h
/home/liu/zephyr-1.14.0-rc1_pthread/samples/hello_world/build/zephyr/include/generated/autoconf.h
zephyr/include/generated/generated_dts_board_fixups.h
zephyr/include/generated/kobj-types-enum.h
```

nslowell, 2019-02-12:
Yup, that's what I'm seeing.

You'll notice this near the bottom of the -H output:

. ../../../include/posix/unistd.h
.. ../../../include/posix/sys/types.h
... ../../../include/posix/sys/types.h  <----------------
.. ../../../include/posix/sys/stat.h

is incorrect and what I believe is the cause of all the trouble.  For some reason the #include_next is looking back at itself instead of finding the sys/types.h in the system directory.  It works correctly with older GCCs.  Still not sure what the problem is or if there's a way to fix it.

closed 2019-02-22 by galak

kelvincheung, 2019-02-23:
@galak 
Did you forget this 'mode_t' define include/posix/mqueue.h?
`typedef unsigned int mode_t;`

When I include both <stdio.h> and <posix/mqueue.h>, the similar error comes out.
Scanning dependencies of target zephyr
[ 15%] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/wifi/api.c.obj
In file included from /home/kelvin/work/zephyr_community/subsys/wifi/os_adapter.h:23,
                 from /home/kelvin/work/zephyr_community/subsys/wifi/api.h:15,
                 from /home/kelvin/work/zephyr_community/subsys/wifi/api.c:16:
/home/kelvin/work/zephyr_community/include/posix/mqueue.h:20:22: error: conflicting types for 'mode_t'
 typedef unsigned int mode_t;
                      ^~~~~~
In file included from /home/kelvin/work/sdk-ng/build/output/arm-zephyr-eabi/arm-zephyr-eabi/include/stdio.h:61,
                 from /home/kelvin/work/zephyr_community/subsys/wifi/os_adapter.h:15,
                 from /home/kelvin/work/zephyr_community/subsys/wifi/api.h:15,
                 from /home/kelvin/work/zephyr_community/subsys/wifi/api.c:16:
/home/kelvin/work/sdk-ng/build/output/arm-zephyr-eabi/arm-zephyr-eabi/include/sys/types.h:205:18: note: previous declaration of 'mode_t' was here
 typedef __mode_t mode_t;  /* permissions */
                  ^~~~~~
make[2]: *** [zephyr/CMakeFiles/zephyr.dir/subsys/wifi/api.c.obj] Error 1
make[1]: *** [zephyr/CMakeFiles/zephyr.dir/all] Error 2
make: *** [all] Error 2

galak, 2019-02-23:
@kelvincheung that should be fixed in #13685

galak, 2019-02-23:
> Yup, that's what I'm seeing.
>
> You'll notice this near the bottom of the -H output:
>
> . ../../../include/posix/unistd.h
> .. ../../../include/posix/sys/types.h
> ... ../../../include/posix/sys/types.h [...]

Was seeing that and thus removed / renamed posix/sys/types.h to posix/posix_types.h

commit e96d0298 (2019-02-22): lib: posix: Fix compile issue with newer newlib

Both SDK 0.10.0-beta2 and the ARM gcc 2018q2 run into a build issue with
newlib and conflict definitions of mode_t type.

First we need to add some ifdef protection if mode_t is already defined
and set _MODE_T_DECLARED if we are the first to define it.

Secondarily, we rename include/posix/sys/types.h to
include/posix/posix_types.h so that we aren't getting a name collusion
with the system sys/types.h and that we can easily and clearily include
it (which we need to do to pull in the info from newlib).

Fixes: #12224