zephyrproject-rtos/zephyr #12225: extractdts should not rename foo-gpios to FOO_GPIO

pabigot, 2018-12-24:
`extract_dts_includes` has:
```
    elif 'pwms' in prop or 'gpios' in prop:
        # drop the 's' from the prop
        generic = prop[:-1]
```

There are 54 DTS yaml entries of the form `foo-gpios`.  All but two (`gpio-keys` and `gpio-leds`) specify:
```
generation: define, use-prop-name
```
`use-prop-name` is required to keep the generation removing the trailing `s` in the property name when synthesizing the `generic` string used for the `#define`.

`use-prop-name` is used 52 times in DTS yaml specifications.  Every single instance is present to override this replacement for `gpios`.

In Linux device tree bindings `foo-gpios` and `pwms` are the standard spelling for these properties.

What was the rationale for changing the generated symbol to be different from the device tree property name?  Is there a good reason why the removal of trailing "s" should be retained?

erwango, 2019-01-02:
@pabigot , this has been introduced in https://github.com/zephyrproject-rtos/zephyr/pull/7197 and #7350. 
When `use-prop-name` was initially introduced, it was not clear this would be the main (and unique) case. The correlation with 's' removal (for which I can't explain the root cause) was a pure code artifact at that time. Code has largely drifted since then, the original option `use-prop-name` is now the main case. If remaining usage is 's' removal, and we're clear it is not needed, it surely can be removed.

pabigot, 2019-01-02:
I've tentatively assigned myself.  It's open for anybody else to take; if that hasn't happened in a couple days I'll work on removal and post a PR if it looks good.