zephyrproject-rtos/zephyr #12226: device tree support gap for SPI chip select

pabigot, 2018-12-24:
It appears Zephyr has incomplete support for SPI devices under device tree: specifically identification of the corresponding chip-select.

AFAICT the sensor drivers that have been converted to device tree for both spi and i2c instances (adxl362, lis2dh) do not support a chip-select GPIO.  lsm6dsl stubs one, but relies on `CONFIG_LSM6DSL_SPI_GPIO_CS` to define it and will never populate it.

In Linux the chip select is provided through the device tree `reg` property, generally as an integer, presumably a globally ordinal pin/GPIO.  In Zephyr's `spi-device.yaml` the `reg` property is declared as type `array`, though I can find nothing that explains what `array` means and how it differs from `compound`.

If Zephyr is intending to use a global integer GPIO index to identify spi chip selects (for bus drivers that don't have dedicated cs signals) there needs to be generic API to do the conversion for each SOC/BOARD.  E.g. Nordic's nrfx HAL will select between GPIO_0 and GPIO_1 based on whether a pin index is greater than 31, but that's below the Zephyr level.

This issue blocks support for the SPI NOR flashes on the Particle development boards as well as nrf52840_pca10056.

vanwinkeljan, 2018-12-25:
Duplicate of #7463, which is assigned to @erwango

pabigot, 2018-12-25:
Thanks for linking, #7463 is definitely relevant.  I don't think this is explicitly duplicative, though, and have revised the description here to more clearly identify the gap and suggest a solution.

galak, 2019-01-09:
I'm not follow what the issue with gpio based CS is.  I know we've had an issue w/regards to supporting multiple devices on the SPI bus from a DTS code generation point of view.  However I'm pretty sure we support at least a single device with its CS controlled by a GPIO.

pabigot, 2019-01-09:
@galak The issue is Zephyr's device-tree YAML specifies that the chip select for a spi-device is identified with an ordinal provided in the `reg` property.  For SoCs/boards that allow arbitrary GPIOs to be used for SPI chip-select signals and that multiple GPIO devices it has to be specified with a gpios compound, preferably named `cs-gpios`.

In Zephyr, the only `cs-gpios` property is at the bus level, not the device level.  The chip-select for a specific device does *not* belong at the bus level when the SOC SPI supports arbitrary GPIOs for chip select, even if there's only one device on the bus in a given hardware configuration.  As I recall DT binding macros don't get generated for things that aren't described in the YAML.

Is that more clear?  If not, then it'll hopefully become clear when a few of my other PRs finally get merged or dropped and I can start providing some of the enhancements that are blocked by them, including ones that require this issue be resolved.

galak, 2019-01-09:
It seems like a few different issues here.  Lets first focus on how this should be described things in DTS first.  We are trying to stay in sync with the general DTS bindings used by Linux.  So the CS should be described in the SPI controller/bus node, not the device.

pabigot, 2019-01-09:
No, I'm not saying remove it from there: it probably has to be there, and probably needs to allow for multiple members, to support SOCs where SPI has a hard-coded set of GPIOs that can be pinmuxed into a  chip select role.  But it also has to be in the device.

galak, 2019-01-09:
> No, I'm not saying remove it from there: it probably has to be there, and probably needs to allow for multiple members, to support SOCs where SPI has a hard-coded set of GPIOs that can be pinmuxed into a chip select role. But it also has to be in the device.

Why does it also need to be in the device?  Why can't the device access cs-gpios in the parent and index it by reg?

pabigot, 2019-01-09:
There's one SPI device in the base, and a second in an overlay.  They have different GPIOs.  How do you specify the set in the bus node?

pabigot, 2019-01-09:
> If you put the overlay aside you'd have:
> 
> `cs-gpios = <&gpiod 13 0>, <&gpioe 0 0>;`

Yes, you would.  Assuming that generates something that can be used in the driver (which is a whole other conversation).  But in the general case you can't put aside the overlay, especially for prototyping where it'll be in the application source rather than a shield.

Another way to handle these type of buses is to go ahead and use the ordinal, leave the bus `cs-gpios` unpopulated, and provide API to convert the ordinal specified by the device to a pair of gpio-device and pin (approach outlined in the initial comment).  That's significantly harder, even if you reject support for overlays that add gpio devices (yes, that is a real use case).

Please don't get hung up on trying to exactly match Linux.  Linux probably doesn't target systems like Nordic nRF5 where the SPI hardware is completely decoupled from managing chip select so any GPIO might have to be supported, including ones that are on an IO extender rather than the SOC.

Regardless, this is a real gap.  I'm happy to delay further bikeshedding until there's something concrete to use as the basis of discussion, or until code generation gets started and we need a resolution, whichever comes first.

galak, 2019-01-09:
> Yes, you would. Assuming that generates something that can be used in the driver (which is a whole other conversation). But in the general case you can't put aside the overlay, especially for prototyping where it'll be in the application source rather than a shield.
> 

I'm not trying to ignore it, I'm trying to see where the real issue is.  Is it a limitation in tooling, in code, in dts itself.  I want us to focus effort on fixing the issue in the right place, and not working around a limitation of a tool.

> Another way to handle these type of buses is to go ahead and use the ordinal, leave the bus `cs-gpios` unpopulated, and provide API to convert the ordinal specified by the device to a pair of [...]

I'm not hung up on it, its just we shouldn't invent something new.  Linux has cases I believe similar to nordic in which the CS management has nothing to do w/the SPI controller.

> Regardless, this is a real gap. I'm happy to delay further bikeshedding until there's something concrete to use as the basis of discussion, or until code generation gets started and we need a resolution, whichever comes first.

Agreed, I think one gap is w/regards to DTC and how overlay's can specify modifications.  I'm thinking we need additional support to manipulate array's.

pabigot, 2019-01-09:
I tend to think it's a limitation in the way Zephyr's using device tree, i.e. through build-time processing producing values that get hard-coded into driver data structures.  It might go away with code generation.  It probably doesn't exist when the drivers can process the tree directly.

At this point I'm not persuaded by the proposal to add array manipulation operations to the device tree source language, but let's see what the upstream response is.

It wouldn't surprise me if Linux has similar SPI cases, I just wasn't able to find them with `git grep` and obvious keywords.  If there's existing practice that we can re-use, it certainly should be preferred.

galak, 2019-01-09:
I'm not sure how any patches or code in Linux will be relevant.  The point is that the DTS in linux support systems where the CS are completely driven by general GPIO pins.  So we will utilize that DTS standard for how we describe CS in Zephyr.

pabigot, 2019-01-10:
I just tried this and found when I change from one `cs-gpios` compound to a second I get these diffs:
```
-#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_CONTROLLER   "GPIO_0"
-#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_FLAGS        0
-#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_PIN          30
+#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_CONTROLLER_0 "GPIO_0"
+#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_CONTROLLER_1 "GPIO_0"
+#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_FLAGS_0      0
+#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_FLAGS_1      0
+#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_PIN_0        30
+#define DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_PIN_1        31

-#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_CONTROLLER   DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_CONTROLLER
-#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_FLAGS        DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_FLAGS
-#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_PIN          DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_PIN
+#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_CONTROLLER_0 DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_CONTROLLER_0
+#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_CONTROLLER_1 DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_CONTROLLER_1
+#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_FLAGS_0      DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_FLAGS_0
+#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_FLAGS_1      DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_FLAGS_1
+#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_PIN_0        DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_PIN_0
+#define DT_NORDIC_NRF_SPI_SPI_1_CS_GPIOS_PIN_1        DT_NORDIC_NRF_SPI_40004000_CS_GPIOS_PIN_1
```
I.e. the name associated with the first instance changed.  I don't think that should happen, since the every drivers would have to support two ways to reference the first controller.

pabigot, 2019-01-23:
Below is from https://github.com/zephyrproject-rtos/zephyr/pull/12647#issuecomment-456769915 which is being fast-tracked.  After further thought, I think the point that we're going to intentionally (eventually) change the one complete stable API in Zephyr based on conformance to the Linux SPI device tree description is worth at least making explicit, if not reconsidering.

> > So I agree w/you that really the APIs around SPI should change and the handling of GPIOs used for CS should be done in the bus controller driver and not specified in each "slave"/client driver. However, I didn't want to take on changing the API at this time.
> 
> I've gotta say, I'm getting a bit of whiplash trying to figure out how Zephyr architectural decisions are made.
> 
> We now agree in #12226 that chip-selects should be associated the bus rather than device because that's what Linux does---even though Zephyr went the other way when the "new" SPI API was added two years ago. That same SPI API which is one of the exactly **two** APIs Zephyr [claims is "stable"](https://github.com/zephyrproject-rtos/zephyr/projects/15) (the other being ADC, which is [not functionally complete](https://github.com/zephyrproject-rtos/zephyr/issues/11922)).
> 
> But instead of either reconsidering that decision and ignoring Linux to put `cs-gpios` in the device where Zephyr expects it, or biting the bullet and changing the API, we're going to have the API and DT architectures remain inconsistent. In an LTS release.