zephyrproject-rtos/zephyr #12228: How to build images for client or server in bluetooth mesh examples?

zzzh, 2018-12-26:
Bluetooth Mesh is supposed to be run in different roles on many devices, right?
If one image is enough, how to set it for different roles?

jhedberg, 2019-01-08:
It's up to you as the application writer to decide what models your node will have. It can have both server and client models (which to some extent are a bit arbitrary since both send and receive messages with the only difference being in who initiates sending request-style messages that the receiver responds to). You could e.g. have a node with buttons mapping to the Generic On/Off Client model and LEDs mapping to the Generic On/Off Server model. Who these models send messages to or receive messages from depends on how you configure your nodes using a Configuration Client Node (often the same device that performs provisioning of your mesh network).

ioannisg, 2019-07-31:
@zzzh please, re-open if needed, to continue discussion

closed 2019-07-31 by ioannisg