zephyrproject-rtos/zephyr #12232: deprecating/removing k_alert API

pabigot, 2018-12-27:
tl;dr: `k_alert` is a fairly large API that has extremely limited use and is easily replaced.  Can we eliminate it?

While looking for a communication mechanism for processing notifications I discovered the [alert API](https://docs.zephyrproject.org/latest/kernel/synchronization/alerts.html).  I then looked for places where it's used. Ignoring its own test suite it's used only for the Intel GNA driver (added two months ago) and a latency benchmark, both of which are easily converted to using a workqueue.

From the logs it appears `k_alert` was designed as an event notification structure, but it wasn't used much.  Prior to its use in intel_gna it was last seen in October 2016 when its use in a timer example was converted to a workqueue (56daa2034566b4209c112860f62cb61627338ce3), possibly as steps towards deprecating it.

Locally I updated intel_gna and the benchmark and removed k_alert completely; sanitycheck passes.

APIs that are no longer used should be deprecated and, as soon as feasible, removed, if for no other reason than to prevent new users from being misled into using them instead of more favored solutions.

Is that the right approach in this case?  What's the process for doing this?  If it's going to be deprecated, it should be marked as such in 1.14-LTS lest it be granted immortality due to new out-of-tree uses.

galak, 2019-01-08:
If we are going to remove this it should be done w/regards to LTS.

andrewboie, 2019-01-10:
I agree, let's remove it completely.

andyross, 2019-01-10:
No complaints here.  An API pruning is absolutely a needed thing, we have a lot of oddball stuff floating around.

pabigot, 2019-01-10:
> @pabigot you have a patch? :-)

The reward for making a suggestion is the opportunity to fix the problem, eh?   Yeah, I made one a couple weeks ago; rebased and checked the updated docs look good.  #12425.

commit bfad9721 (2019-01-17): kernel: remove k_alert API

This API was used in only one place in non-test code.  See whether we
can remove it.

Closes #12232
closed 2019-01-17 by nashif