zephyrproject-rtos/zephyr #12241: logging: log output busy loops if log output function can not process buffer

b0661, 2018-12-29:
**Describe the bug**
According to the documentation the  log_output_func shall return the number of bytes processed.

```
typedef int (*log_output_func_t)(u8_t *buf, size_t size, void *ctx)

    Prototype of the function processing output data.

    Return
        Number of bytes processed. 
    Parameters

            data: Data.
            length: Data length.
            ctx: User context.
```

If the function returns 0 the log buffer_write function busy loops forever.

```
static void buffer_write(log_output_func_t outf, u8_t *buf, size_t len,
			 void *ctx)
{
	int processed;

	do {
		processed = outf(buf, len, ctx);
		len -= processed;
		buf += processed;
	} while (len);
}
```

**To Reproduce**
Steps to reproduce the behavior:
Change the log_output_func to return 0.

**Expected behavior**
Ether - The logger shall stop the output to the backend and come back later.
Or - Improve documentation to reflect the needed behaviour (which is still unclear to me)

**Impact**
Application stalls if log output stalls.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
 - Zephyr, Commit SHA 3980e0cf642cd1d31b56bc8e7c15a99592548fd1

**Additional context**
N/A

nordic-krch, 2019-01-07:
@b0661 processed means dropped as well. It is backend responsibility to drop data. I would prefer to keep log_output module simple. It's just a helper module for backends to avoid code repetition. I will add a note to api documentation, ok?

b0661, 2019-01-07:
@nordic-krch 

> It's just a helper module for backends to avoid code repetition. I will add a note to api documentation, ok?

Can you just make it **typedef void (*log_output_func_t)(u8_t *buf, size_t size, void *ctx)**? 

By this it wouid be clear by interface that the whole buffer has to be handled by the backend. You may get an extra loop in the backend but could get rid of e.g. the buffer_write helper.

Otherwise a note in the api documentation is also fine for me.

carlescufi, 2019-01-22:
@nordic-krch can you submit a PR for the note in the doc that you agreed to?

PR #14919 (2019-03-27): logging: Clarify contract of log_output_func_t
The log_output_func_t backend function is supposed to either process or
drop bytes and return the amount of those to the caller. Clarify this in
the documentation.

Fixes #12241
commit 082cf7ac (2019-03-27): logging: Clarify contract of log_output_func_t

The log_output_func_t backend function is supposed to either process or
drop bytes and return the amount of those to the caller. Clarify this in
the documentation.

Fixes #12241
closed 2019-03-27 by carlescufi