zephyrproject-rtos/zephyr #12243: Build error with I2C driver for nucleo_f103rb: DT_ST_STM32_I2C_V1_40005400_BASE_ADDRESS' undeclared here

mattzzw, 2018-12-30:
**Describe the bug**
Trying to compile master (65f79cdc2) with nucleo_f103rb and I2C_1

Results in 

```
In file included from /Users/mattzz/Code/zephyr/include/arch/arm/arch.h:20,
                 from /Users/mattzz/Code/zephyr/include/arch/cpu.h:15,
                 from /Users/mattzz/Code/zephyr/include/kernel_includes.h:34,
                 from /Users/mattzz/Code/zephyr/include/kernel.h:17,
                 from /Users/mattzz/Code/zephyr/include/device.h:11,
                 from /Users/mattzz/Code/zephyr/include/clock_control.h:14,
                 from /Users/mattzz/Code/zephyr/include/drivers/clock_control/stm32_clock_control.h:12,
                 from /Users/mattzz/Code/zephyr/drivers/i2c/i2c_ll_stm32.c:8:
zephyr/include/generated/generated_dts_board.h:296:45: error: 'DT_ST_STM32_I2C_V1_40005400_BASE_ADDRESS' undeclared here (not in a function); did you mean 'DT_ST_STM32_USART_40004400_BASE_ADDRESS'?
 #define DT_I2C_1_BASE_ADDRESS               DT_ST_STM32_I2C_V1_40005400_BASE_ADDRESS
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/generated/generated_dts_board.h:296:45: note: in definition of macro 'DT_I2C_1_BASE_ADDRESS'
 #define DT_I2C_1_BASE_ADDRESS               DT_ST_STM32_I2C_V1_40005400_BASE_ADDRESS
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/generated/generated_dts_board.h:303:31: error: 'DT_ST_STM32_I2C_V1_40005400_CLOCK_BITS' undeclared here (not in a function); did you mean 'DT_ST_STM32_USART_40004400_CLOCK_BITS'?
 #define DT_I2C_1_CLOCK_BITS   DT_ST_STM32_I2C_V1_40005400_CLOCK_BITS
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/generated/generated_dts_board.h:303:31: note: in definition of macro 'DT_I2C_1_CLOCK_BITS'
 #define DT_I2C_1_CLOCK_BITS   DT_ST_STM32_I2C_V1_40005400_CLOCK_BITS
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/generated/generated_dts_board.h:304:30: error: 'DT_ST_STM32_I2C_V1_40005400_CLOCK_BUS' undeclared here (not in a function); did you mean 'DT_ST_STM32_USART_40004400_CLOCK_BUS'?
 #define DT_I2C_1_CLOCK_BUS   DT_ST_STM32_I2C_V1_40005400_CLOCK_BUS
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/generated/generated_dts_board.h:304:30: note: in definition of macro 'DT_I2C_1_CLOCK_BUS'
 #define DT_I2C_1_CLOCK_BUS   DT_ST_STM32_I2C_V1_40005400_CLOCK_BUS
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/generated/generated_dts_board.h:302:45: error: 'DT_ST_STM32_I2C_V1_40005400_CLOCK_FREQUENCY' undeclared here (not in a function); did you mean 'DT_ST_STM32_USART_40004400_CLOCK_BUS'?
 #define DT_I2C_1_BITRATE                    DT_ST_STM32_I2C_V1_40005400_CLOCK_FREQUENCY
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/generated/generated_dts_board.h:302:45: note: in definition of macro 'DT_I2C_1_BITRATE'
 #define DT_I2C_1_BITRATE                    DT_ST_STM32_I2C_V1_40005400_CLOCK_FREQUENCY
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/mattzz/Code/zephyr/include/clock_control.h:14,
                 from /Users/mattzz/Code/zephyr/include/drivers/clock_control/stm32_clock_control.h:12,
                 from /Users/mattzz/Code/zephyr/drivers/i2c/i2c_ll_stm32.c:8:
zephyr/include/generated/generated_dts_board.h:299:49: error: 'DT_ST_STM32_I2C_V1_40005400_LABEL' undeclared here (not in a function); did you mean 'DT_ST_STM32_USART_40004400_LABEL'?
 #define CONFIG_I2C_1_NAME                       DT_ST_STM32_I2C_V1_40005400_LABEL
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/mattzz/Code/zephyr/include/device.h:107:11: note: in definition of macro 'DEVICE_AND_API_INIT'
   .name = drv_name, .init = (init_fn),     \
           ^~~~~~~~
/Users/mattzz/Code/zephyr/drivers/i2c/i2c_ll_stm32.c:238:34: note: in expansion of macro 'CONFIG_I2C_1_NAME'
 DEVICE_AND_API_INIT(i2c_stm32_1, CONFIG_I2C_1_NAME, &i2c_stm32_init,
                                  ^~~~~~~~~~~~~~~~~
[35/115] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work_q.c.obj
ninja: build stopped: subcommand failed.
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_f103rb ..
3. ninja menuconfig (see below)
4. ninja

```
$ grep I2C zephyr/.config
# CONFIG_I2C_STM32_INTERRUPT is not set
CONFIG_I2C_STM32=y
CONFIG_I2C_STM32_V1=y
CONFIG_HAS_DTS_I2C=y
CONFIG_I2C=y
# CONFIG_I2C_SLAVE is not set
# CONFIG_I2C_GPIO is not set
# CONFIG_I2C_SBCON is not set
CONFIG_I2C_INIT_PRIORITY=60
# CONFIG_I2C_0 is not set
CONFIG_I2C_1=y
# CONFIG_I2C_2 is not set
# CONFIG_I2C_3 is not set
# CONFIG_I2C_4 is not set
# CONFIG_I2C_5 is not set
# CONFIG_I2C_6 is not set
# CONFIG_I2C_7 is not set
CONFIG_USE_STM32_LL_I2C=y
```

My understanding is that I have to use I2C_1 according to `stm32f1.dtsi`

```
		i2c1: i2c@40005400 {
			compatible = "st,stm32-i2c-v1";
			clock-frequency = <I2C_BITRATE_STANDARD>;
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40005400 0x400>;
			clocks = <&rcc STM32_CLOCK_BUS_APB1 0x00200000>;
			interrupts = <31 0>, <32 0>;
			interrupt-names = "event", "error";
			status = "disabled";
			label= "I2C_1";
```

**Expected behavior**
Compile w/o error

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain (e.g Zephyr SDK, ...)

```
$ cmake -GNinja -DBOARD=nucleo_f103rb ..
Zephyr version: 1.13.99
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.2", minimum required is "3.4")
-- Selected BOARD nucleo_f103rb
Parsing Kconfig tree in /Users/mattzz/Code/zephyr/Kconfig
Loading /Users/mattzz/Code/zephyr/boards/arm/nucleo_f103rb/nucleo_f103rb_defconfig as base
Configuration written to '/Users/mattzz/Code/zph_test/build/zephyr/.config'
-- Loading /Users/mattzz/Code/zephyr/boards/arm/nucleo_f103rb/nucleo_f103rb.dts as base
-- Overlaying /Users/mattzz/Code/zephyr/dts/common/common.dts
-- Cache files will be written to: /Users/mattzz/Library/Caches/zephyr
-- The C compiler identification is GNU 8.2.1
-- The CXX compiler identification is GNU 8.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mattzz/Code/zph_test/build
-- Commit SHA or Version used
```

**Additional context**
I'm not very familiar with DTS but I can't find the file that defines e.g. `DT_ST_STM32_I2C_V1_40005400_BASE_ADDRESS`. What am I missing?

ydamigos, 2018-12-30:
I2C_1  isn't enabled in nucleo_f103rb dts file. Try to add the following in nucleo_f103rb dts file.

```
&i2c1 {
	status = "ok";
	clock-frequency = <I2C_BITRATE_FAST>;
};
```
Please consider creating a PR if it works.

erwango, 2019-01-03:
Changing this from Bug to Enhancement as this is a matter of correct board configuration.

nashif, 2019-01-11:
> Changing this from Bug to Enhancement as this is a matter of correct board configuration.

does not really qualify as an enhancement, it is a bug :)

erwango, 2019-01-14:
@nashif, well I2C1 is not configured on this board, and actually I2C driver has not been developed/validated on this STM32F1 series. Build issue happens because @mattzzw tries to enable it without adding the requested matching fixup definition.
Since I2C was not supported before, current work in enhancement rather than fix.

closed 2019-01-14 by erwango

mattzzw, 2019-01-14:
It's also working on nucleo_f103rb with PR #12246.