zephyrproject-rtos/zephyr #12245: LWM2M registration timeout

arn-soft, 2018-12-30:
**Describe the bug**
The lwm2m example throws **LWM2M registration timeout** . The device seems to connect to the leshan server as i can see that on web interface of leshan-demo-server (which is running on [http://localhost:8080](http://localhost:8080) ). the client last update changes in every 30-45 second with new registration id.
**uart log:** 
[00:04:52.172,454] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'nrf52840_pca10056' and client lifetime 30
[00:04:52.682,037] <dbg> net_lwm2m_rd_client.sm_send_registration: Non dtls
[00:04:52.682,067] <dbg> net_lwm2m_rd_client.sm_send_registration: registration sent [fdde:ad00:beef:0:3696:3458:e373:be67]
[00:04:55.036,499] <dbg> net_lwm2m_engine.retransmit_request: Resending message: 0x20001750
[00:04:59.726,745] <dbg> net_lwm2m_engine.retransmit_request: Resending message: 0x20001750
[00:05:09.097,259] <dbg> net_lwm2m_engine.retransmit_request: Resending message: 0x20001750
[00:05:27.828,308] <wrn> net_lwm2m_rd_client: Registration Timeout
[00:05:27.828,308] <dbg> net_lwm2m_client_app.rd_client_event: Disconnected
[00:05:27.828,308] <dbg> net_lwm2m_client_app.rd_client_event: Registration failure!

**To Reproduce**
Steps to reproduce the behavior:
1. cd sample/net/lwm2m
2. mkdir build && cd build
3. cmake -GNinja -DBOARD=nrf52840_pca10056 -DOVERLAY_CONFIG=overlay-ot.conf ..
[overlay-ot.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/2717127/overlay-ot.conf.txt)

4. ninja

**Expected behavior**
It should successfully connected to the leshan server

**Screenshots or console output**
[image]
[image]

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr-sdk
 - Commit id : 06d7835

mike-scott, 2019-01-04:
Oddly, I don't see overlay-ot.conf here: https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/net/lwm2m_client

I was going to add one, once https://github.com/zephyrproject-rtos/zephyr/pull/12343 is merged and communication via OT border router confirmed.

But right now, I don't think this is an issue?

arn-soft, 2019-01-05:
I am using overlay-ot.conf defined in echo_client example. I think it is serving it's purpose. please take a look at look if i am missing something in that file. and of course i will check that PR #12343

nashif, 2019-02-11:
@mike-scott what is the status here?

mike-scott, 2019-02-11:
I need to merge an OpenThread overlay.conf that I've tested before this can be fixed.  It's not so much of a "bug" as an untested feature atm.

mike-scott, 2019-02-11:
@nashif Do I have the ability to get an OpenThread overlay for LwM2M into 1.14 now that the freeze is active?

mike-scott, 2019-02-11:
Removed bug tag and replaced with feature

arn-soft, 2019-03-08:
got it working after pr merger.

closed 2019-03-08 by arn-soft

andynnd, 2019-08-22:
Hi @arn-soft, Can you help to share your overlay-ot.conf file for lwm2m. I test with the overlay-ot.conf.txt you shared in the first post but it can not run. Thanks.

mike-scott, 2019-08-22:
@andynnd Can you open a new issue asking for an overlay-ot.conf to be added to the LwM2M sample?  I have one and will get to it when I'm back from ELC.

andynnd, 2019-08-23:
@mike-scott I just open a new issue at https://github.com/zephyrproject-rtos/zephyr/issues/18601. Thanks.