zephyrproject-rtos/zephyr #12249: Can't flash CC3220 with OpenOCD

Andreasdahlberg, 2018-12-31:
**Describe the bug**
I tried flashing the hello world sample application but the flash command returns an error(see console output) and the application is not loaded onto the board.

I have also tried loading the bin-file using UniFlash with no success. UniFlash does not report any errors but the application does not appear to be running(not console output at all).

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=cc3220sf_launchxl ..
3. ninja flash
4. See error

**Expected behavior**
I expect the hello world application to be loaded into flash.

**Impact**
Showstopper since i can't test any applications on target.

**Screenshots or console output**
```
[1/1] Flashing cc3220sf_launchxl
Using runner: openocd
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst
adapter_nsrst_delay: 1100
cortex_m reset_config vectreset
adapter speed: 2500 kHz
Info : XDS110: connected
Info : XDS110: firmware version = 2.3.0.16
Info : XDS110: hardware version = 0x0027
Info : XDS110: connected to target via SWD
Info : XDS110: SWCLK set to 2500 kHz
Info : clock speed 2500 kHz
Info : SWD DPIDR 0x2ba01477
Info : cc32xx.cpu: hardware has 6 breakpoints, 4 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc32xx.cpu         cortex_m   little cc32xx.dap         running
Info : SWD DPIDR 0x2ba01477
Error: timed out while waiting for target halted
target did not halt at reset vector
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x0101a9c8 psp: 0x200018a0
auto erase enabled
wrote 14336 bytes from file /home/andreas/Programmering/zephyr/samples/hello_world/build/zephyr/zephyr.elf in 0.383157s (36.539 KiB/s)
Error: DP initialisation failed
in procedure 'reset' 
in procedure 'ocd_bouncer'

Info : XDS110: disconnected
ERROR: command exited with status 1: /usr/local/bin/openocd -s /usr/local/bin/openocd/scripts -f /home/andreas/Programmering/zephyr/boards/arm/cc3220sf_launchxl/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/andreas/Programmering/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'reset halt' -c 'verify_image /home/andreas/Programmering/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'reset run' -c shutdown
run as "west -v ... flash ..." for a stack trace
```

**Environment:**
 - OS: Ubuntu 16.04
 - Toolchain: Zephyr SDK
 - Commit SHA: 65f79cdc21783ab376b80657136788ad479a6c28

**Additional context**
I have put the board into “Development Mode” using UniFlash and jumper SOP(j15) is set to [001].

I have also checked that CONFIG_XIP=y is set.

vanti, 2019-01-03:
Which version of openocd are you using?

I am working on Ubuntu 18.04 and am on the last commit of the 'develop' branch, and can confirm that 'ninja flash' works for me:

```
$ ninja flash
[0/1] Flashing cc3220sf_launchxl
Using runner: openocd
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst
adapter_nsrst_delay: 1100
cortex_m reset_config vectreset
adapter speed: 2500 kHz
Info : XDS110: connected
Info : XDS110: firmware version = 2.3.0.16
Info : XDS110: hardware version = 0x0027
Info : XDS110: connected to target via SWD
Info : XDS110: SWCLK set to 2500 kHz
Info : clock speed 2500 kHz
Info : SWD DPIDR 0x2ba01477
Info : cc32xx.cpu: hardware has 6 breakpoints, 4 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc32xx.cpu         cortex_m   little cc32xx.dap         running
Info : SWD DPIDR 0x2ba01477
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted
target did not halt at reset vector
target halted due to debug-request, current mode: Handler SysTick
xPSR: 0x6100000f pc: 0x010012e8 msp: 0x20000e38
auto erase enabled
wrote 14336 bytes from file /home/zephyr-work/samples/hello_world/build/zephyr/zephyr.elf in 1.411737s (9.917 KiB/s)
Info : SWD DPIDR 0x2ba01477
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted
target did not halt at reset vector
target halted due to debug-request, current mode: Handler SysTick
xPSR: 0x6100000f pc: 0x01001014 msp: 0x20000e38
verified 13628 bytes in 0.565471s (23.535 KiB/s)
Info : SWD DPIDR 0x2ba01477
shutdown command invoked
Info : XDS110: disconnected
```

If you had tried to use Uniflash to flash the .bin file, please first revert by removing the MCU
Image (.bin) file from Uniflash and reflashing the device with just the other files, as per step 3 in the Prerequisites section in https://docs.zephyrproject.org/latest/boards/arm/cc3220sf_launchxl/doc/cc3220sf_launchxl.html. Then go back to running 'ninja flash' and see if it helps.

Best regards,
Vincent

Andreasdahlberg, 2019-01-03:
It's working now, i tried your suggestion about reflashing the device without the MCU Image and it solved the issue.

Thank you for the help!

Best regards,
Andreas

closed 2019-01-03 by Andreasdahlberg