zephyrproject-rtos/zephyr #12252: potential non-termination of k_mem_pool_alloc

LVPGroup, 2019-01-02:
When I call k_mem_pool_alloc by passing the "size_t size" larger than "max_sz" and "s32_t timeout" is K_FOREVER, _sys_mem_pool_block_alloc always returns "-ENOMEM" and the while(1) loop in k_mem_pool_alloc does not terminate.

andyross, 2019-01-02:
That's easy enough to handle but requires an extra test in every allocation.  I mean, our blocking allocator API is sort of unique, other mallocs would just return failure synchronously, so it's not like there's a surprising-behavior argument here.  Are we sure we don't want this (allocating a block that can never fit) to be a "user error" kind of thing?

Alternatively: there are lots of situations where the app can create a situation where an allocation will block forever (e.g. just allocate all the memory) in ways we obviously can't fix, and we'd consider those application bugs.  Why is this case **not** an application bug?

Third way to think about it: when you make a blocking call with a timeout of K_FOREVER, you need to be prepared for what happens when you indeed wait forever.

Bottom line is that I'm not convinced this is worth handling as a special case, but I'm willing to be overruled.

andrewboie, 2019-01-17:
I agree with @andyross this seems like a garbage-in-garbage-out situation.

andrewboie, 2019-01-17:
Closing this; not seeing the advantage to adding this check when the parameters are clearly incorrect and K_FOREVER is used

closed 2019-01-17 by andrewboie