zephyrproject-rtos/zephyr #12257: The latest GNU ARM Embedded Toolchain can't produce hex files on Windows

SebastianBoe, 2019-01-02:
**Describe the bug**
The latest release of "GNU ARM Embedded Toolchain" fails when trying to create a hex file on the Windows platform.

**To Reproduce**
1. Build a sample that creates a hex file (for instance the hello world sample with nrf52_pca10040) with the environment described below.
2. See error

**Expected behavior**
No build failure.

**Actual behavior**
Error:

arm-none-eabi-objcopy.exe: zephyr.hex 64-bit address 0x4b4fa30004000 out of range for Intel Hex file

**Impact**
Unable to use the latest release of gnuarmemb on Windows.

**Environment (please complete the following information):**
1. Only affects Windows
2. Use exactly GNU Arm Embedded Toolchain **8-2018-q4-major**

**Additional context**
This has been independently reported upstream by another user:

https://bugs.launchpad.net/gcc-arm-embedded/+bug/1810274

closed 2019-03-15 by galak

gokcedilek, 2019-06-23:
Hello,
I am trying to compile and debug my code in VS Code for the STM32 F401RE MCU. When I attempt to build my project, I am getting the following error: "C:/Program Files (x86)/GNU Tools ARM Embedded/8 2018-q4-major/bin/arm-none-eabi-objcopy:build/vscode2.hex: bad value". What would you suggest doing in this case?
Thank you very much!

E3V3A, 2019-09-04:
This issue and the PR's related to this, are very confusing, to say the least. 

* Has this actually been fixed?
* Can we still use GNU ARM Embedded to compile or do we need to change?

SebastianBoe, 2019-09-04:
This was fixed with the release of

GNU Arm Embedded Toolchain Version 8-2019-q3-update Released: July 10, 2019

E3V3A, 2019-09-04:
@SebastianBoe
Since you seem to be from Trondheim, you may be working with Nordic Semi. If that is so, I would like you to tell them to update all their *`getting started`* and *`setup`* pages, which still refer users to use the older tool-chain before this bug...

SebastianBoe, 2019-09-04:
Thank you for the heads up, will pass it along.