zephyrproject-rtos/zephyr #12272: SD/MMC interface support

pic16f887, 2019-01-03:
I have started with generic SD/MMC interface + STM32F7 series platform specific driver for Zephyr RTOS. 
Could you assign this issue to me?

easonxiang, 2019-01-07:
@pic16f887 We are planing to support MMC with Zephyr system,  I was wondering when the generic SD/MMC interface will be ready, do you have a schedule for this feature?

Thanks.

pic16f887, 2019-01-07:
@easonxiang initial version of SD/MMC interface will be ready approximately in two-three weeks.

easonxiang, 2019-01-08:
@pic16f887 Thanks for your response, looking forward to see it.

yashi, 2019-01-29:
I'm also interested in SD/MMC support.  There is another issue about supporting SDHC, #5457. And SPI based #5528, which is already merged as #11793.

pic16f887, 2019-01-30:
Currently I am testing my code. This week or probably next week I will create a pull request with my code. The code supports only SD/MMC HW interface (no SPI). Initially without DMA support. DMA will be added later.

ChaserTlss, 2019-04-08:
I'm interested in this features,  Any update in this patch? Why this patch is not merge.
What's the new plan for this patch?

hjuul, 2020-02-27:
Seems this initiative is abandoned. I need this feature for stm32f7 myself, and I consider basing it off of the code from @pic16f887.  Or does someone know of any other ongoing work on SD Memory Card for Zephyr?

erwango, 2022-06-08:
supported

closed 2022-06-08 by erwango