zephyrproject-rtos/zephyr #12315: [Coverity CID :190618]Code maintainability issues in /drivers/modem/wncm14a2a.c

mandarcthorat1, 2019-01-04:
Static code scan issues seen in File: /drivers/modem/wncm14a2a.c
 Category: Code maintainability issues
 Function: wncm14a2a_rx
 Component: Drivers
 CID: 190618
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nashif, 2019-03-04:
```
returned_value: Assigning value from net_buf_findcrlf(rx_buf, &frag, &offset) to len here, but that stored value is overwritten before it can be used.
```

mike-scott, 2019-03-04:
Hmm, I should probably ignore the return value entirely there, and remove the check for !frag.  Either way we need to break the loop because a handler was found.

closed 2019-03-07 by galak