zephyrproject-rtos/zephyr #12319: [Coverity CID :190614]Control flow issues in /subsys/shell/shell_utils.c

mandarcthorat1, 2019-01-04:
Static code scan issues seen in File: /subsys/shell/shell_utils.c
 Category: Control flow issues
 Function: buffer_trim
 Component: Other
 CID: 190614
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

jarz-nordic, 2019-01-08:
Sorry I can't accesss it. I have following error:

> Error code 8
> The proxy failed to connect to the web server, due to TCP connection rejection (TCP Reset).

carlescufi, 2019-01-08:
I get the same. Here we go again with the Coverity issues. Let's bring it up with the TSC.

jarz-nordic, 2019-01-22:
@mandarcthorat1: Can you share some details about this bug? I am still not able to access your link.

mandarcthorat1, 2019-01-24:
Hi @jarz-nordic ,

The issue is not related to access rights. The coverity website is currently down.

maxxlife, 2019-02-10:
It seems that coverity website is down more than a half of the month. I have the same problem with your links.

maxxlife, 2019-02-17:
I suppose that issue is fixed in commit f7f4fe33a12e07d2c992012aee7e2832c9214356

maxxlife, 2019-02-21:
Working on that issue. Tomorrow I will finish it.

jarz-nordic, 2019-02-22:
@maxxlife : but what is the issue?

maxxlife, 2019-02-22:
According to the coverity report and my tests it is a dead code part in
function buff_trim.

maxxlife, 2019-02-24:
@jarz-nordic Hi. I checked it more carefully and it seems that code without dead code part. It works well. No need to fix it. We should just provide comments to square it off.

PR #13720 (2019-02-26): subsys: Changed dead code part in function buffer_trim()
According to the Coverity issue 190614 (github issue #12319)
Fix for logically dead code in function buffer_trim
Before while execution cannot reach this statement: buff[0] = '\0';

Tested and noticed that part of code is unnecessary inside of while
For the test, I made the same function in CodeBlocks and send to it
string to detect and delete whitespaces. After deletion of dead code
part, I tested function again, it works.

Function buffer_trim first detects whitespace characters in the end
of the string using first while, then it adds  [... cut]
closed 2019-02-26 by nashif