zephyrproject-rtos/zephyr #12321: [Coverity CID :190612]Control flow issues in /subsys/net/lib/lwm2m/lwm2m_rw_json.c

mandarcthorat1, 2019-01-04:
Static code scan issues seen in File: /subsys/net/lib/lwm2m/lwm2m_rw_json.c
 Category: Control flow issues
 Function: json_add_char
 Component: Networking
 CID: 190612
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

maxxlife, 2019-02-09:
Hi @mandarcthorat1 , I'm trying to solve your issue, but link is not working. Do you what happened? 
Service Unavailable

The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.
Apache/2.4.29 (Ubuntu) Server at scan9.coverity.com Port 443

mike-scott, 2019-02-13:
Fixed in merged commit: https://github.com/zephyrproject-rtos/zephyr/commit/181544beec6b20991593672491a05af9efec2fb0#diff-061d8c2c19cda9a1150de0483f46984b

Can close.

closed 2019-02-13 by mike-scott