zephyrproject-rtos/zephyr #12322: [Coverity CID :190611]Control flow issues in /drivers/usb/device/usb_dc_nrfx.c

mandarcthorat1, 2019-01-04:
Static code scan issues seen in File: /drivers/usb/device/usb_dc_nrfx.c
 Category: Control flow issues
 Function: usb_dc_detach
 Component: Drivers
 CID: 190611
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

maxxlife, 2019-02-09:
Hi, I'm trying to solve your issue, but link is not working. Do you know what happened? 
Service Unavailable

The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.
Apache/2.4.29 (Ubuntu) Server at scan9.coverity.com Port 443

Jocelyn-Li, 2019-02-12:
@maxxlife It is working now. Log in using your github account and request Zephyr project access.

maxxlife, 2019-02-12:
I already made it on the Coverity page.

maxxlife, 2019-02-17:
Hi. @Jocelyn-Li I think that issue has already fixed in merged commit 85d75ec853ada2cfdc38fea6b592de04965614c6

nashif, 2019-03-04:
fixed.

closed 2019-03-04 by nashif