zephyrproject-rtos/zephyr #12359: Default address selection for IPv6 should follow RFC 6724

mike-scott, 2019-01-07:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, Zephyr uses a simple diff function to determine the IPv6 source address for outbound data.  In many cases this isn't correct (read as: any use case that has more than 1 non-link local unicast address).  As a result, OpenThread routing via a border router for internet communication is broken.

**Describe the solution you'd like**
RFC 6724 (https://tools.ietf.org/html/rfc6724) describes the correct process for selecting the default source address.

**Describe alternatives you've considered**
I brought up a solution for OpenThread source address, and the RFC was mentioned as the correct solution.

**Additional context**
none

mike-scott, 2019-01-07:
This was brought up as a discussion in https://github.com/zephyrproject-rtos/zephyr/pull/12343

mike-scott, 2019-01-07:
NOTE that OpenThread adds a couple of rules:

- Don't use addresses returning true for IsAnycastRoutingLocator()
- Add rule: Prefer EID

jukkar, 2020-01-28:
Going through old issues. This is a good candidate for closing as the address selection seems to work ok atm. Can be re-opened if really needed.

closed 2020-01-28 by jukkar