zephyrproject-rtos/zephyr #12369: WDT: wdt callbacks are not getting triggerred before CPU going for a reboot

ravindranaththiyagarajan, 2019-01-08:
@kl-cruz @carlescufi 

In NRF52840, we are registering the wdt callbacks with zephyr as shown below in the code snippet.

Expected behavior:
---------------------------

These registered callbacks should executed on watchdog time out.
As per the NRF52840 wdt specification, The watchdog should generate an interrupt on timeout event. The CPU reset is postponed for two 32.768KHz clock cycles.

Current behavior:
-------------------------
1. We are seeing the watchdog interrupt nrfx_wdt_irq_handler not getting triggered and the registered callbacks never run, but CPU reboots.
2. We are not sure if the CPU is getting rebooted much before the wdt irq is fired.

Code snippet:
--------------------

    struct wdt_timeout_cfg wdt_config;
    system_wdt_dev = device_get_binding(CONFIG_WDT_0_NAME);
    // Reset SoC when watchdog timer expires.
    wdt_config.flags = WDT_FLAG_RESET_SOC;
    // Expire watchdog after x milliseconds.
    wdt_config.window.min = 0;
    wdt_config.window.max = 5000
    // Set up watchdog callback. Jump into it when watchdog expired.
    wdt_config.callback = system_wdt_callback;
    system_wdt_channel_id = wdt_install_timeout(system_wdt_dev, &wdt_config);

carlescufi, 2019-01-08:
@anangl @kl-cruz can you please take a look at this one?

kl-cruz, 2019-01-08:
sure, I'll take care on it!

ravindranaththiyagarajan, 2019-01-15:
With this flag WDT_OPT_PAUSE_HALTED_BY_DBG, I am able to the stop the program control in the interrupt handler with out CPU getting reset before.

closed 2019-01-15 by ravindranaththiyagarajan