zephyrproject-rtos/zephyr #12386: Support USB host on Zephyr

bublover, 2019-01-09:
**Is your feature request related to a problem? Please describe.**
We're working for UNISOC and doing upstream for new chip and board.
https://github.com/zephyrproject-rtos/zephyr/pull/10626
Now we need to support pen drive which is working on USB host mode.
but there is USB device mode only on Zephyr.

**Describe the solution you'd like**
Please provide a framework of USB host, like driver, api, protocol, etc..

**Describe alternatives you've considered**
Please provide thinking and methods that I might use to add.

**Additional context**
NA

JuniorJPDJ, 2020-02-09:
I'm also interested in this feature, and as far as I've found in internet - there are much more requests for usb host support ;/

raveslave, 2021-01-22:
checking in if any plans from NXP to contribute with host support for LPC or i.MX

MaureenHelm, 2021-01-22:
> checking in if any plans from NXP to contribute with host support for LPC or i.MX

No plans yet

jfischer-no, 2021-01-22:
@efdev Please add EHCI driver to #30361 even if it was not quite finished.

kevin0402, 2021-05-05:
> > checking in if any plans from NXP to contribute with host support for LPC or i.MX
> 
> No plans yet

@MaureenHelm  
Can we get any help from NXP if we want to port USB host stack on i.MX RT1060 platform ?
The stack is based on #30361

MaureenHelm, 2021-05-07:
> Can we get any help from NXP if we want to port USB host stack on i.MX RT1060 platform ?
> The stack is based on #30361

@kevin0402 I don't have a developer available to work on this right now, but if you want to drive the effort, we can help if you have questions. PRs are welcome.

raveslave, 2022-04-04:
> > checking in if any plans from NXP to contribute with host support for LPC or i.MX
> 
> No plans yet

how about now? :-)

MaureenHelm, 2022-04-04:
> > > checking in if any plans from NXP to contribute with host support for LPC or i.MX
> > 
> > 
> > No plans yet
> 
> how about now? :-)

I'm no longer in a position to speak for NXP. I defer to @dleach02

DRNadler, 2022-11-24:
@dleach02 Any update on USB host and pull-request above? We'd love to use Zephyr but cannot without USB host (as many others)... Thanks!

raveslave, 2024-02-12:
both nxp's lpc and mcx support host, as well as upcoming nordics. joint effort?

DRNadler, 2024-02-13:
We cannot use Zephyr for several projects until USB Host mode and ideally OTG are supported. Lots of other firms are in the same boat...

jfischer-no, 2024-02-15:
Progress on USB support is covered in https://github.com/zephyrproject-rtos/zephyr/issues/42066. There is no need for another opaque enhancement issue.

closed 2024-02-15 by jfischer-no