zephyrproject-rtos/zephyr #12391: Http server post handling

chulbumi, 2019-01-09:
I am testing an HTTP server sample on QEMU.
I sent this post request.

curl -d '{"key1":"value1", "key2":"value2"}' -H "Content-Type: application/json" -X POST 192.0.2.1

----------
this is output from QEMU.

[00:00:33.263,171] <inf> net_app: Connection from 192.0.2.2:53028 (0x00401310)
[00:00:33.266,584] <dbg> net_http.http_received: (0x00401bd8): [0x004004c0] Received 160 bytes http data
[00:00:33.269,808] <dbg> net_http._http_change_state: (0x00401bd8): [0x004004c0] state CLOSED (0) => WAITING_HEADER (1) [on_url():954]
[00:00:33.271,520] <dbg> net_http._http_change_state: (0x00401bd8): [0x004004c0] state WAITING_HEADER (1) => RECEIVING HEADER (2) [on_header_field():922]
[00:00:33.277,642] <dbg> net_http._http_change_state: (0x00401bd8): [0x004004c0] state RECEIVING HEADER (2) => OPEN (4) [http_process_recv():512]
[00:00:33.278,517] <inf> net_http: [0x004004c0] HTTP connection from  (0x00401310)
[00:00:33.279,165] <dbg> net_http_server_sample.http_connected: HTTP connect attempt URL ▒U@
[00:00:33.279,655] <dbg> net_http_server_sample.http_serve_index_html: Sending index.html (170 bytes) to client
[00:00:33.284,145] <dbg> net_http.http_send_msg_raw: (0x00401bd8): [0x004004c0] Sending 253 bytes data
[00:00:33.303,532] <dbg> net_http_server_sample.http_sent: Index sent
[00:00:33.317,410] <dbg> net_http_server_sample.http_sent: Index sent
[00:00:33.323,386] <dbg> net_http._http_change_state: (0x00401bd8): [0x004004c0] state OPEN (4) => CLOSED (0) [http_closed():530]
[00:00:33.323,720] <dbg> net_http.http_closed: (0x00401bd8): [0x004004c0] http closed
[00:00:33.323,861] <dbg> net_http_server_sample.http_closed: Connection 0x004004c0 closed
----------
There is http_received callback on this sample.
But there is no output message like "Received %d bytes data".
I have no idea how I can handle this post message body.
Could you give me a hint for this?

jukkar, 2019-01-10:
Note that the sample HTTP server is extremely simple, you need to enhance it to process your POST request.

jukkar, 2019-01-15:
The sample server is just a sample, it would need changes to handle POST request. There are no such plans atm, it is up to the user to create such a server that fits user needs.

closed 2019-01-15 by jukkar