zephyrproject-rtos/zephyr #12395: Bluetooth samples wont run using the latest branch

summer20100514, 2019-01-09:
**Describe the bug**
most of the samples of bluetooth won't run as expected compared with the older version

**To Reproduce**
mesh demos, mesh onoff, mesh onoff_level, peripheral and so on

**Expected behavior**
None

**Impact**
None

**Screenshots or console output**
None

**Environment (please complete the following information):**
 - OS: Windows
 - Commit SHA 3e06b7cfa690c94efec69ebb1690f664baccbeb3

simply put, has zephyr tested the samples inside sample/bluetooth at all? it seems that no tests are performed before released

cvinayak, 2019-01-10:
@summer20100514 The changes merged to master is manually sanity tested with few samples, specially the samples/bluetooth/peripheral. I have tested the master commit (2 days old) 4e23d6c87594080f85760e92e50860f8ddbdbe5e and do not see any regression related to Bluetooth interoperability with phones. Please elaborate your observations because I cannot find the your mentioned SHA in the Zephyr repository.

jhedberg, 2019-01-10:
I just did a quick test with the samples/boards/reel_board/mesh_badge app from latest master, and it works fine: both the initial peripheral (GATT server) functionality as well as the mesh functionality that it switches to after writing a name to it over GATT. I did this for two reel boards and they were able to send mesh messages to each other just fine. I also can't find the commit that's referenced in the PR description.

summer20100514, 2019-01-11:
have you test mesh related app such as onoff_level_lighting_vnd_app ?

jhedberg, 2019-01-11:
> have you test mesh related app such as onoff_level_lighting_vnd_app?

@summer20100514 samples/boards/reel_board/mesh_badge *is* a mesh related app. I didn't try the onoff related ones since a reel board was at hand but an nRF52 DK wasn't (at least immediately)

Could you also please clarify your commit SHA reference in the PR description? Where did you take that from, since there's no such commit in the upstream master branch?

summer20100514, 2019-01-11:
Sorry, I merged the master branch into my own. The SHA actually is c922ce6ce45173c2b1f2c8c48f57f275d9a046d8. Nonetheless, what I have added is just some debugging logs.

summer20100514, 2019-01-11:
I just wonder why there is always MPU FAULT messages. I have checked the stack size.
samples\boards\nrf52\mesh\onoff-app\
> Bluetooth initialized
> Mesh initialized
> ***** MPU FAULT *****
>   Instruction Access Violation
> ***** Hardware exception *****
> Current thread ID = 0x20000c3c
> Faulting instruction address = 0x4f480688
> Fatal fault in thread 0x20000c3c! Aborting.

summer20100514, 2019-01-11:
@jhedberg current thread id above is adv_thread_data

jhedberg, 2019-01-11:
@summer20100514 could you try increasing ADV_STACK_SIZE in subsys/bluetooth/host/mesh/adv.c? Set it to e.g. 1024. If the issue doesn't go away it might be some other thread that overflows writes into the adv thread's stack.

summer20100514, 2019-01-11:
@jhedberg hi, I have changed the ADV_STACK SIZE to 1024, but the MPU FAULT is the same.

jhedberg, 2019-01-14:
@summer20100514 no, since I can't reproduce this

jhedberg, 2019-01-14:
@summer20100514 is it really so that absolutely no Bluetooth samples work for you? Or is it only the Mesh-related samples? You asked me about those, but you've also mentioned "peripheral" in your description.

summer20100514, 2019-01-14:
@jhedberg @cvinayak Sorry for the delay. I finally found out the reason for crashing is that I try to access one uninitialized pointer, and I feel very embarrassed.

jhedberg, 2019-01-14:
@summer20100514 no worries, this kind of stuff happens to everyone :) So it's ok to close this issue now?

summer20100514, 2019-01-15:
@jhedberg yes, be kind to do so.

closed 2019-01-15 by jhedberg