zephyrproject-rtos/zephyr #12405: add test to catch issues fixed in Fix kernel #12384

nashif, 2019-01-09:
Add a test that would catch the issue fixed in #12384

nashif, 2021-05-28:
@LeiW000  @enjiamai  do we have test coverage for this?

enjiamai, 2021-05-28:
@nashif  ,  do you mean which test case cover it?
[image]

enjiamai, 2021-05-28:
Hi @nashif  we have the poll tests that cover this, I think it could be closed.

nashif, 2021-05-28:
> Hi @nashif we have the poll tests that cover this, I think it could be closed.

This is not exactly test coverage. The issue was that is_polling was set to true, and the test still passed, see the original commit in #12384

enjiamai, 2021-05-28:
> > Hi @nashif we have the poll tests that cover this, I think it could be closed.
> 
> This is not exactly test coverage. The issue was that is_polling was set to true, and the test still passed, see the original commit in #12384

Got your point. After checking, it cannot be caught by the current testcases. Ok, we will do it soon.

cfriedt, 2021-07-14:
@LeiW000 , @Zhaoningx - any progress on this?

LeiW000, 2021-07-14:
@Zhaoningx is going to work on it next week. Now he is busy on another urgent issue.

Zhaoningx, 2021-08-03:
Hi @nashif , the code changed a lot after #12384, and I spent some time to understand if the code `poller.is_polling = false` makes sense. The code as below:
[image]

The bool value is_polling is used to judge the event is existing or not in k_poll function, in the function signal_poll_event, it seems it's not working to set the poller value, because if a event is fulfilled successfully, the event will be reset to NULL,also reset is_polling, if failed, a negative value will be returned and the thread of called k_poll function will don't callback. So @nashif , do you think the code of signal_poll_event make sense?

Zhaoningx, 2021-08-04:
Hi @nashif , the code changed a lot after PR #12384, and I spent some time on this, and try to design testcase to detect is_polling value. But didn't work, in my opinion, it seems the is_polling is used in the main function k_poll to decide whether need to register an event and polling it, the value of is_polling that we want to detect is in signal_poll_event function, as you know, this function is used to fulfill an event and wake up k_poll thread callback. If successful, the event will be set to NULL whatever is_polling is true or false. If failed, it means the thread of k_poll expired, and k_poll thread will not wake up and callback. It seems the is_polling in signal_poll_event function can control nothing. I don't know how is_polling of signal_poll_event can control whether to poll and how it can be detected. So I would like to ask @nashif for help to comment here. Thanks.

Zhaoningx, 2021-08-05:
Hi @nashif , could you take a look at this, thanks very much.

PR #37744 (2021-08-19): Add a testcase to detect is_polling
Add a testcase to detect is_polling, to avoid some
issues.

Fixed #12405
commit c0d18079 (2021-08-19): poll: add a testcase to detect is_polling

Add a testcase to detect is_polling, to avoid some
issues.

Fixed #12405
closed 2021-08-19 by cfriedt