zephyrproject-rtos/zephyr #12409: non-tickless kernels incorrectly advance system clock with delayed ticks

pabigot, 2019-01-10:
In the discussion of #12332 it was determined that the timer updates around 2018-09 through 2018-10 resulted in an unintentional change of behavior for `CONFIG_TICKLESS_KERNEL=n`.

Prior to the rework the system clock advanced by one tick on each interrupt, regardless of whether the interrupt was serviced late.

Support for `CONFIG_TICKLESS_KERNEL=y` inadvertently changed this so that the system clock was advanced by the number of ticks required to align it with the divided hardware clock, regardless of the value of `CONFIG_TICKLESS_KERNEL`.

The timer implementations need to be reworked so that for `CONFIG_TICKLESS_KERNEL=n` the original behavior is restored.  Note that this will result in an non-linear relation between the hardware and system clocks if the tick interrupt is delayed by at least `sys_clock_hw_cycles_per_tick()`.

pabigot, 2019-01-10:
The initial issue description summarizes my understanding of the issue, which may be incorrect.  I've opened the issue because the resolution to #12332 will involve a revision of the documentation for `k_timer` that must refer to the difference between tickless and non-tickless timer behavior.  The content of the documentation depends on whether this bug is fixed.

carlescufi, 2019-01-22:
@andyross is this something you plan to address in the near future?

carlescufi, 2019-01-22:
Originally mentioned in [here](https://github.com/zephyrproject-rtos/zephyr/issues/12332#issuecomment-452807555)

andyross, 2019-01-23:
Do we consider it a bug?  This is the way Zephyr has behaved forever and the existing design just preserves the behavior.

Quite frankly I'm sort of exhausted with timer at this point.  Tell me what needs fixing and I'll fix it.  I have no horse in this particular race.

pabigot, 2019-01-23:
@andyross I understood you to say it was a bug (well, an unintentional change) here: https://github.com/zephyrproject-rtos/zephyr/issues/12332#issuecomment-452807555

I see you later commented you didn't feel this was important, so feel free to close this to confirm that position.  I only opened it because the only record of the behavior change is buried in comments in a relatively unrelated issue, and it's hard enough to find out the why of past changes without having to wade through issues with dozens of comments.

andyross, 2019-02-28:
Cleaning up low hanging bugs in prep for release, finally got to this one

PR #13962 (2019-03-01): drivers/timer: Restore non-tickless tick count behavior
The newer series of timer drivers will compare counters vs. the last
tick boundary to compute a number of ticks to announce to the kernel.
In the case of CONFIG_TICKLESS=n, this actually represents a change of
behavior from our old scheme where "ticks" always reflected the number
of interrupts received.

The distinction only matters when an interrupt is delayed more than a
full tick, of course.  But that actually makes a difference to some
timekeeping code.  Restore the old behavior.

This also has the benefit of further reducing code siz [... cut]
commit 7cbdb6c5 (2019-03-01): drivers/timer: Restore non-tickless tick count behavior

The newer series of timer drivers will compare counters vs. the last
tick boundary to compute a number of ticks to announce to the kernel.
In the case of CONFIG_TICKLESS=n, this actually represents a change of
behavior from our old scheme where "ticks" always reflected the number
of interrupts received.

The distinction only matters when an interrupt is delayed more than a
full tick, of course.  But that actually makes a difference to some
timekeeping code.  Restore the old behavior.

This also has the benefit of further reducing code si [... cut]
closed 2019-03-01 by andrewboie