zephyrproject-rtos/zephyr #12419: cannot flash with segger jlink in windows environment

DenvE, 2019-01-10:
**Describe the bug**
When running zephyr on windows and using the cmake/ninja tools, you can not use segger jlink to flash your image.

problem 1: the script mentioned jlinkexe ( which is the exe file on linux/max, on windows it is jlink.exe)
problem 2: jlink.exe is also a java binary ( C:\Program Files\Java\jdk-11.0.1\bin\jlink.exe ), so be sure to add FILEPATH ( like it is done for openocd)

**To Reproduce**
Steps to reproduce the behavior:
1. set ZEPHYR_TOOLCHAIN_VARIANT and  GNUARMEMB_TOOLCHAIN_PATH according your development board
2. cd sample\helloworld
3. mkdir build; cd build
4. cmake -GNinja -DBOARD=stm32f746g_disco -DSTLINK_FW=jlink ..
5. ninja
6. ninja flash

**Expected behavior**
jlink on windows should work.

**Impact**
Created local work-around

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: windows
 - Zephyr 1.13

**Additional context**
Add any other context about the problem here.

carlescufi, 2019-01-10:
Please add the JLink folder to your PATH. This is typically located in:
`C:\Program Files (x86)\SEGGER\JLink_V634e`
replacing the version of course.
The test if you can execute `jlink` from the command prompt and you get the JLink interface.

Also you will need this to be merged first: https://github.com/zephyrproject-rtos/zephyr/pull/12380

closed 2019-01-11 by carlescufi

DenvE, 2019-01-11:
it works but still having problems when jlink.exe ( the java linker ) is available in the environment.
could using FILEPATH  be a solution?

DenvE, 2019-01-11:
just discovered that FILEPATH is filled in when running host-tools.cmake
adding this

# jlink is an optional dependency
find_program(
  JLINK
  jlink
  )

gives CMakeCache.txt

//Path to a program.
JLINK:FILEPATH=C:/Program Files/Java/jdk-11.0.1/bin/jlink.exe

PR #12380 (2019-01-11): Update west to ad6a22c
This brings in a new runner.
Fixes #12419