zephyrproject-rtos/zephyr #12432: Support multi USB classes simultaneously

RamiroOliveira, 2019-01-11:
**Is your enhancement proposal related to a problem? Please describe.**
At the moment in Zephyr it's only possible to assign one type of each USB class to a device

**Describe the solution you'd like**
An simple method to had more than one of each USB class to a device. For example, I wanted to create a USB device with two ttyACMs/ttyUSB. With the current implementation this is not possible.

**Describe alternatives you've considered**
If this is not in the immediate plans an Application Note describing the general steps would be interesting.

@finikorg

masz-nordic, 2019-01-11:
Relates to #12060

carlescufi, 2019-01-14:
CC @masz-nordic @jfischer-phytec-iot @pawelzadrozniak

carlescufi, 2019-01-14:
@finikorg As part of this work it would be great to have an exmaple that combines 2 different classes each multiple instances, eg: 2x CDC ACM ports and 2x HID instances

PR #13373 (2019-03-05): usb: add hid-cdc example
This example combines 2 HID classes and 2 CDC ACM classes to create a composite, multi-instance device.

Closes #12432.
closed 2019-03-05 by carlescufi